flipperdevices/flipperzero-firmware #2872: Flipper crashing on HardFault

anarchopossum, 2023-07-12:
### Describe the bug.

Flipper keeps crashing when left alone or just by using it.

### Reproduction

1. switch on
2. wait for 10-15 minutes
3. crash screen shows
4. reboot

### Target

0.86.1 Firmware

### Logs

```Text
18 [default] Binding on background is not deferred as requested by the DeferredPropertyNames class info because one or more of its sub-objects contain an id.
18 [default] Binding on contentItem is not deferred as requested by the DeferredPropertyNames class info because one or more of its sub-objects contain an id.
19 [default] Binding on animation is not deferred as requested by the DeferredPropertyNames class info because one or more of its sub-objects contain an id.
20 [default] Binding on background is not deferred as requested by the DeferredPropertyNames class info because one or more of its sub-objects contain an id.
20 [default] Binding on contentItem is not deferred as requested by the DeferredPropertyNames class info because one or more of its sub-objects contain an id.
21 [default] Binding on background is not deferred as requested by the DeferredPropertyNames class info because one or more of its sub-objects contain an id.
22 [default] Binding on contentItem is not deferred as requested by the DeferredPropertyNames class info because one or more of its sub-objects contain an id.
22 [default] Binding on background is not deferred as requested by the DeferredPropertyNames class info because one or more of its sub-objects contain an id.
25 [default] Binding on background is not deferred as requested by the DeferredPropertyNames class info because one or more of its sub-objects contain an id.
25 [default] Binding on contentItem is not deferred as requested by the DeferredPropertyNames class info because one or more of its sub-objects contain an id.
25 [default] Binding on indicator is not deferred as requested by the DeferredPropertyNames class info because one or more of its sub-objects contain an id.
25 [default] Binding on background is not deferred as requested by the DeferredPropertyNames class info because one or more of its sub-objects contain an id.
136 [APP] qFlipper version 1.3.2 commit 7b0839e1 2023-06-08T03:58:22
136 [APP] OS info: macOS 13.2 13.2 22.3.0 Qt 6.3.1
163 [REG] Detected new device: VID_0x483:PID_0x5740
206 [DBG] Trying serial port  at /dev/cu.wlan-debug
206 [DBG] Trying serial port  at /dev/tty.wlan-debug
206 [DBG] Trying serial port  at /dev/cu.Bluetooth-Incoming-Port
206 [DBG] Trying serial port  at /dev/tty.Bluetooth-Incoming-Port
206 [DBG] Trying serial port flip_Invadh at /dev/cu.usbmodemflip_Invadh1
206 [DBG] Using  serial port flip_Invadh at /dev/cu.usbmodemflip_Invadh1
212 [RPC] Starting RPC session...
308 [RPC] RPC session started successfully.
308 [RPC] (1) System Protobuf Version START
366 [DBG] Detected protobuf version: 0.18
366 [RPC] (1) System Protobuf Version SUCCESS
366 [RPC] (2) Property Get START
488 [RPC] (2) Property Get SUCCESS
488 [RPC] (3) Storage Info @/ext START
540 [RPC] (3) Storage Info @/ext SUCCESS
540 [RPC] (4) Storage Stat @/ext/Manifest START
608 [RPC] (4) Storage Stat @/ext/Manifest SUCCESS
609 [RPC] (5) System Get DateTime START
661 [DBG] Flipper time skew is -698 milliseconds
661 [RPC] (5) System Get DateTime SUCCESS
661 [RPC] (6) System Set DateTime START
684 [UPD] Fetched update information from https://update.flipperzero.one/qFlipper/directory.json
713 [RPC] (6) System Set DateTime SUCCESS
713 [RPC] Stopping RPC session...
716 [RPC] RPC session stopped successfully.
716 [REG] Registering the device
716 [DEV] Version: 0.86.1 commit: 3991db4b radio: 1.15.0
716 [BKD] Current device changed to "Invadh"
779 [UPD] Fetched update information from https://update.flipperzero.one/firmware/directory.json
842 [RPC] Starting RPC session...
945 [RPC] RPC session started successfully.
945 [RPC] (1) Gui Start ScreenStream START
1007 [RPC] (1) Gui Start ScreenStream SUCCESS
28810 [RPC] (2) Storage Info @/ext START
28874 [RPC] (2) Storage Info @/ext SUCCESS
28877 [RPC] (3) Gui Stop ScreenStream START
28940 [RPC] (3) Gui Stop ScreenStream SUCCESS
[... 110 lines cut ...]
37511644 [RPC] (3) Storage Info @/ext START
37511698 [RPC] (3) Storage Info @/ext SUCCESS
37511698 [RPC] (4) Storage Stat @/ext/Manifest START
37511757 [RPC] (4) Storage Stat @/ext/Manifest SUCCESS
37511757 [RPC] (5) System Get DateTime START
37511817 [DBG] Flipper time skew is -1854 milliseconds
37511817 [RPC] (5) System Get DateTime SUCCESS
37511817 [RPC] (6) System Set DateTime START
37511874 [RPC] (6) System Set DateTime SUCCESS
37511874 [RPC] Stopping RPC session...
37511877 [RPC] RPC session stopped successfully.
37511878 [REG] Registering the device
37511878 [DEV] Version: 0.86.1 commit: 3991db4b radio: 1.15.0
37511878 [BKD] Current device changed to "Invadh"
37512035 [RPC] Starting RPC session...
37512133 [RPC] RPC session started successfully.
37512134 [RPC] (1) Gui Start ScreenStream START
37512195 [RPC] (1) Gui Start ScreenStream SUCCESS
39455944 [UPD] Fetched update information from https://update.flipperzero.one/firmware/directory.json
39498777 [UPD] Fetched update information from https://update.flipperzero.one/qFlipper/directory.json
```

### Anything else?

Screen says: Flipper crashed and was rebooted ST(R) Corpro(R) HardFault

skotopes, 2023-08-01:
Yep, known issue. We are waiting for new Core2 firmware from ST.
Meanwhile please switch Settings->System->Sleep Method to Legacy.

closed 2023-08-01 by skotopes

rbn369, 2024-07-04:
Hi - how did you get past that first error in log 18 -25? I'm stuck there...thanks! :)

skotopes, 2024-07-06:
@rbn369 which error? this specific issue(crash in deep sleep mode) was fixed long time ago. Open new issue and properly describe your problem