flipperdevices/flipperzero-firmware #2877: U2F Security Key not supported in Windows 11

SystemOfTheCSGO, 2023-07-14:
### Describe the enhancement you're suggesting.

The current module of security key U2F is not supported by windows.

Reproduction

Use the firmware as usual, get a clean install of windows 11, try to link the security key from flipper to windows, you will get a error: this device is not certified.

### Anything else?

This can be bypassed emulating headers of a certified key. (I’ve seen it done before, at least a homemade microsoft certified’ seckey.)

kamilskrzypek4, 2023-07-19:
The key for MS products like Windows, O365 needs to be FIDO2, as far as I know flipper is FIDO1. Any chance for FIDO2 on flipper hardware?

zhiyan114, 2023-07-19:
> The key for MS products like Windows, O365 needs to be FIDO2, as far as I know flipper is FIDO1. Any chance for FIDO2 on flipper hardware?

I feel like it would be a security risk, especially since FIDO2 is usually a passwordless login. Definitely safer to get a separate security key for it.

skotopes, 2023-08-01:
We'll try to pack it in future updates.

SolarSciencePup, 2023-09-24:
> We'll try to pack it in future updates.

Any updates?

skotopes, 2023-09-26:
Not really, plans are to complete NFC refactoring, then BadUSB/BadUART, and then U2F/FIDO.

Mechcondrid, 2024-01-10:
Any news? this would help me out since now i need this for U2F on work repos

FastLionArt, 2024-02-22:
Hey there. I can help with developing U2F/FIDO :)

alexh3791, 2024-02-28:
I just wanted to chime in and say that I'm really looking forward to FIDO2 on the Flipper as well. 

Adding NFC support to the U2F/FIDO app would be extremely helpful, too, since I authenticate from an iPhone more often than a laptop/desktop at this point.

AngelFluffyOokami, 2024-07-24:
> Not really, plans are to complete NFC refactoring, then BadUSB/BadUART, and then U2F/FIDO.

From what I see searching around the repo, NFC refactoring has been completed, I may be wrong and/or misinterpreted, I'm not sure about whether BadUSB/BadUART has been, but if so, is FIDO2 support still in the pipeline?

skotopes, 2024-07-24:
In a long term FIDO2 is on the roadmap. But right now no one is working on it and we also'd like to have proper secure enclave before going further.

exussum12, 2025-05-06:
https://github.com/polhenarejos/pico-fido

That is compatible with an ESP32 Not sure what would be needed to get STM32 support  but posibly part way there.

This spoofs VID too gives better compatibility

Luk164, 2025-07-22:
Has there been any news on FIDO2 support lately?

exussum12, 2026-05-01:
Would prefer someone on the project to look, Looking at it, I think its possible but not easy :(