flipperdevices/flipperzero-firmware #2879: custom frequencies in setting_user is read by subghz app but not reflected in GUI (flipper)

cyb3rjerry, 2023-07-14:
### Describe the bug.

Hey there! So after poking around I've realized there seem to be some clash between the subghz frequencies you can read (that's displayed in the GUI) and what's actually being read by the app.
Here is the content of my file:
```
# to use manual settings and prevent them from being deleted on upgrade, rename *_user.example files to *_user
Filetype: Flipper SubGhz Setting File
Version: 1
# Add Standard frequencies included with firmware and place user frequencies after them
Add_standard_frequencies: true

# Default Frequency: used as default for "Read" and "Read Raw"
Default_frequency: 433920000

# Frequencies used for "Read", "Read Raw" and "Frequency Analyzer"
Frequency: 300000000
Frequency: 310000000
Frequency: 320000000
Frequency: 419000000

# Frequencies used for hopping mode (keep this list small or flipper will miss signal)
Hopper_frequency: 300000000
#Hopper_frequency: 310000000
#Hopper_frequency: 310000000

# Custom preset
# format for CC1101 "Custom_preset_data:" XX YY XX YY .. 00 00 ZZ ZZ ZZ ZZ ZZ ZZ ZZ ZZ, where: XX-register, YY - register data, 00 00 - end load register, ZZ - 8 byte Pa table register

#Custom_preset_name: AM_1
#Custom_preset_module: CC1101
#Custom_preset_data: 02 0D 03 07 08 32 0B 06 14 00 13 00 12 30 11 32 10 17 18 18 19 18 1D 91 1C 00 1B 07 20 FB 22 11 21 B6 00 00 00 C0 00 00 00 00 00 00

#Custom_preset_name: AM_2
#Custom_preset_module: CC1101
#Custom_preset_data: 02 0D 03 07 08 32 0B 06 14 00 13 00 12 30 11 32 10 17 18 18 19 18 1D 91 1C 00 1B 07 20 FB 22 11 21 B6 00 00 00 C0 00 00 00 00 00 00
```
The file is placed under `SD Card/subghz/assets/` 
[image]

Here's a vid where we can see the frequencies of the setting_user file aren't shown in the UI

https://github.com/flipperdevices/flipperzero-firmware/assets/92945113/0cc54f09-7642-431d-9dc1-4f5de2b4d6d7

I've also attached below the logs showing the app is loading the frequencies

### Reproduction

1. Place the setting_user file content above under `SD Card/subghz/assets/setting_user`
2. Go to the subghz app
3. Enter the `read` command
4. Go to the config
5. Scroll through the frequencies to see that the "custom" ones are missing

### Target

Firmware: 0.86.2 Build date: 11 Jul 2023 Hardware: 12.F7B9C6 Radio fw: 1.15.0LITE

### Logs

```Text
Firmware version: 0.86.2 0.86.2 (c29941ae built on 12-07-2023)

>: log
Press CTRL+C to stop...
6690234 [I][Loader] App returned: 0
6690236 [I][Loader] Application stopped. Free heap: 141840
6690664 [I][AnimationManager] Select 'L1_Mods_128x64' animation
6690668 [I][AnimationManager] Load animation 'L1_Mods_128x64'
6690697 [I][SavedStruct] Loading "/int/.desktop.settings"
6692571 [I][Loader] Starting Sub-GHz
6692578 [I][AnimationManager] Unload animation 'L1_Mods_128x64'
6692600 [I][SubGhzSetting] Keeping standard frequencies
6692609 [I][SubGhzSetting] Frequency loaded 300000000
6692613 [I][SubGhzSetting] Frequency loaded 310000000
6692618 [I][SubGhzSetting] Frequency loaded 320000000
6692621 [I][SubGhzSetting] Frequency loaded 419000000
6692640 [I][SubGhzSetting] Hopper frequency loaded 300000000
6692666 [I][SubGhzKeystore] Loading keystore /ext/subghz/assets/keeloq_mfcodes
6692719 [I][SubGhzKeystore] Loading keystore /ext/subghz/assets/keeloq_mfcodes_user
6692725 [E][SubGhzKeystore] Unable to open file for read: /ext/subghz/assets/keeloq_mfcodes_ur
6692727 [I][SubGhzProtocoNiceFlorS] Loading rainbow table from /ext/subghz/assets/nice_flor_s
6692731 [I][SubGhzProtocoCameAtomo] Loading rainbow table from /ext/subghz/assets/came_atomo
6692734 [I][SubGhzProtocoAlutech_at_4n] Loading rainbow table from /ext/subghz/assets/alutechn
6692786 [I][elf] Total size of loaded sections: 4340
6692789 [I][SubGhzDeviceRegistry] Loaded 2 radio device
6692916 [I][SubGhz_Device_CC1101_Ext] Init OK
6693042 [I][SubGhz_Device_CC1101_Ext] Init OK
```

Skorpionm, 2023-07-15:
try scrolling all the way to the right.  new ones are added to the end of the list

cyb3rjerry, 2023-07-15:
> try scrolling all the way to the right.  new ones are added to the end of the list

That was exactly it 🤦‍♂️I'm not familiar with the codebase yet but I'm guessing the values are simply appended to the end of the list rather than inserted at the right place?

closed 2023-07-15 by cyb3rjerry