flipperdevices/flipperzero-firmware #2885: Guidance for using furi_hal_crypto*

twisted-pear, 2023-07-17:
### Description of the feature you're suggesting.

Hi,

this feature request is about extending both the documentation and the capabilities of the functionality exposed in furi_hal_crypto.h.

I was wondering about how developers of FAPs can safely utilize furi_hal_crypto.h. From what I can see there are functions to add new keys to the key store and to use them to encrypt or decrypt in CBC mode. However, while there are 100 key slots it's not possible to just pick a key slot to store a key in. Instead you need to pick the next key slot that is still free.

This poses some problems for developing an application that wants to use its own keys because it's not predictable which key slot the application will be able to use.

Say for example we have app A and app B. If the user uses A first, it will install its key into key slot 12 (the first 10 are factory keys and slot 11 is used by the u2f app). So then app B will get slot 13. However, if the user uses the apps in the reverse order, app B will get slot 12 and app A will get slot 13.

My requested feature is threefold:

- Please provide documentation on the keys slots that are already in use by default. This should contain which keys are in which slots, what they are used for and whether they are unique per device or the same across devices. If this documentation already exists then I was unable to find it. (Please point me to it.) Obviously some of the keys are considered secret, but it should at least be possible to know what they are used for and if they are unique per device.
- Please provide some guidance to app developers on how to add keys so that multiple apps that use key slots can coexist. How should developers keep track of key slots? Should each app using a key slot maintain a config file on the SD card that stores the key slot in use? Is using these key slots discouraged entirely?
- In the future it might also be helpful to provide developers with some kind of mapping system that can map an app ID to one or more key slots. That way it's not necessary for individual apps to keep track of the key slots they are using.

Thank you and keep up the good work!

Skorpionm, 2023-07-22:
the first 5 keys are the same across all devices, the next 5 are unique.  adding keys to the vault is a one way operation and there is no option to erase 1 of the keys, only all of them can be deleted.  in this case, the already installed keys will be lost, I would not recommend to abuse the addition of new keys, but, if possible, use the existing ones

twisted-pear, 2023-07-22:
If 5 keys are unique per device, how does furi_hal_crypto_verify_enclave() work? For each of the ten keys it encrypts some test vectors with a static IV and compares that to a static output. If some keys are unique per device that shouldn't work, since the test vectors, IVs and expected outputs are hard-coded.

Would it be possible to use the hardware encryption with keys provided by the software and not supplied by the enclave?

akopachov, 2023-07-24:
>the first 5 keys are the same across all devices

Is there anybody who can explain the purpose of having "keys" which are the same across all the devices? Asking this because, I guess, nobody will argue against the fact that such a "keys" can not be considered as a keys and instead should be considered as an absence of a key. And if so, then what is the reason to have them at all?

skotopes, 2023-07-26:
@twisted-pear @akopachov Statement by @Skorpionm that second 5 keys are unique is incorrect.

There are 11 keys in total provisioned at factory:
- 0 - master key, meant for key provisioning in encrypted form, can not be used directly
- 1-10 - 10 keys that flipper firmware uses internally or going to use in future

Slots 11-100 are user keys. Those key slots are explicitly left empty to guarantee that we don't know them. There are 2 ways to generate and store keys in those slots:
- Provision them manually right after receiving device(using `crypto` cli command or API)
- Use `furi_hal_crypto_verify_key` to check specific user slot and generate random key if it doesn't exists. (Examples can be found in `u2f_data_cert_key_load` function. U2F application uses slot 11. Also `furi_hal_crypto_verify_key` function name is misleading, it's going to be renamed in future update.)

Please wait till the next release where we finalize API and provide examples with firmware.

skotopes, 2023-09-20:
I think it's safe to close this ticket. Let me know if I anything else is needed.

closed 2023-09-20 by skotopes