flipperdevices/flipperzero-firmware #2906: Bluetooth Devices Scanner

Fishrock123, 2023-07-20:
### Description of the feature you're suggesting.

It would be cool if Flipper Zero could scan for bluetooth devices and show detailed info.

I have a bluetooth device which I think may be malfunctioning since I can't detect it from an iPhone but I think flipper could do more, detect if there is any signal / information even if it is incorrect, for debugging other devices.

skotopes, 2023-08-01:
That is possible with full radio stack. It requires additional space in flash memory which we don't have. So currently it is impossible.

We are currently talking with ST about that, we may return to this feature at some point in future.

closed 2023-08-01 by skotopes

Fishrock123, 2023-08-01:
What is ST? Is it possible to make this as an application that can be downloaded onto the SD card from the list of applications?

skotopes, 2023-08-01:
@Fishrock123 this one https://www.st.com/content/st_com/en.html
nope it's not application it's core2 firmware.

jkramarz, 2024-12-03:
@skotopes, is BLE Central role still missing in the firmware?

skotopes, 2024-12-04:
@jkramarz there is not enough memory(both ram and flash) for full stack.

elhenro, 2026-10-30:
this is possible with custom firmware: https://github.com/elhenro/flipper-ble-scanner