flipperdevices/flipperzero-firmware #2908: Option to disable emotions

adryd325, 2023-07-21:
### Description of the feature you're suggesting.

An option to disable emotions other than happy would be appreciated. I legitimately feel bad whenever I don't use my flipper for a while because of the angry mode

skotopes, 2023-08-01:
That is a part of the character ;-)

closed 2023-08-01 by skotopes

domints, 2024-09-10:
It's also unnecessarily wearing out EEPROM. It was fun for the first year, not it's more annoying than anything :) I'd love to skip on emotions and leveling. Just make flipper flip :)

portasynthinca3, 2024-09-10:
By they way, we added an optional happy mode in #3863 (part of the 1.0 firmware release)

domints, 2024-09-10:
Just got information from my friends about firmware 1.0, and that's great! But I looked at the PR and it looks like it didn't stop writing status info to eeprom, just ignores what's in there, doesn't it?