flipperdevices/flipperzero-firmware #2914: Flipper crashing on infrared remote loading

minchogaydarov, 2023-07-22:
### Describe the bug.

Flipper crashes with "furi_check failed" when trying to use infrared remote.

### Reproduction

1. Select the IR remote
2. When opening it the Flipper crashes

[image]
[image]
[Daikin FTXC35DV1B Full.ir.zip](https://github.com/flipperdevices/flipperzero-firmware/files/12136713/Daikin.FTXC35DV1B.Full.ir.zip)

skotopes, 2023-08-02:
Basically it runs our of memory because of remote size. We'll change how we load big files in future updates.

skotopes, 2023-08-02:
It's going to take some time because of required refactoring. Meanwhile you can split remote into multiple files.

skotopes, 2024-01-10:
Was fixed some time ago. Latest release should work without issues.

closed 2024-01-10 by skotopes