flipperdevices/flipperzero-firmware #2925: Stealth mode

Zarcolio, 2023-07-28:
### Description of the feature you're suggesting.

Currently, it's easy for EDR to detect a Flipper Zero. It would be really nice if a stealth mode could be configured in which USB and Bluetooth connections can set to use custom VID & PID and MAC addresses. This way EDR detection could hopefully be bypassed.

skotopes, 2023-08-01:
What is EDR?

Zarcolio, 2023-08-01:
EDR is Endpoint Detection and Response, it can be seen as the evolution of antivirus software. 
I.e. Microsoft Defender for Endpoint 
For more information, see https://www.gartner.com/reviews/market/endpoint-detection-and-response-solutions

uhlhosting, 2024-01-03:
How is the progress here?

skotopes, 2024-01-03:
In backlog, will return to it after BLE

melayyalem, 2024-06-20:
Helpful article for this: https://blog.grumpygoose.io/hunting-flipper-zero-db260274c45c
Also bad_kb_app is using default values for VID/HID but not setting the manufacturer name

GottZ, 2024-11-08:
[image]
essentially this is what the monitoring emails receive. I suspect it's just usb vendor and class id matching.