flipperdevices/flipperzero-firmware #2926: Share NFC/RFID Key to be emulated on Mobile

DarkEnder4583, 2023-07-28:
### Description of the feature you're suggesting.

I would like to request a feature that would allow for NFC/RFID Keys to be saved to a paired mobile device to be emulated later from said mobile device (whether in built in “wallet” applications, or from the Flipper app itself).
I believe this would allow for a much wider range of usage for these two protocols.

skotopes, 2023-07-31:
That is already implemented

closed 2023-07-31 by skotopes

DarkEnder4583, 2023-07-31:
Meaning, you can use the phones own NFC/RFID emulator? Would you mind telling me how? 
I'm aware of the ability to remotely have the flipper emulate saved keys from itself, if that is what you are referring too.

skotopes, 2023-08-01:
Oh, you meant emulating on the phone itself. That is quite bad idea in general: phone NFC capabilities are severely limited, in the best case they support NFC-A emulation. Also this questions should be asked in https://github.com/flipperdevices/Flipper-Android-App/issues

DarkEnder4583, 2023-08-01:
Oh, gotcha. Thanks!

And apologies for posting this in the firmware repository; I tried filing it in the iOS App repository, but it didn't allow me. I assumed the Android repository would have been the same way, but I appear to be wrong 😅 Thanks again!