flipperdevices/flipperzero-firmware #2927: Gen4 authentication in NFC Magic app

sfjuocekr, 2023-07-30:
### Describe the bug.

The NFC Magic app has issues doing the authentication with Gen4.

I have a Gen4 card with config block: 0000DEADBEEF02000978009102DABC1910101112131415160400080089F8
The used password is: DEADBEEF

I try to authenticate this card using the Flipper Zero and it comes up with "not magic or unsupported card".

When I issue a raw command: CFDEADBEEFC6
The output is correct: 0000DEADBEEF02000978009102DABC1910101112131415160400080089F8 [A35C690B]

The trace output shows:

```
      Start |        End | Src | Data (! denotes parity error)                                           | CRC | Annotation
------------+------------+-----+-------------------------------------------------------------------------+-----+--------------------
          0 |        992 | Rdr |40                                                                       |     | 
    1054432 |    1055424 | Rdr |52                                                                       |     | 
    1056692 |    1057972 | Tag |04                                                                       |     | 
    1094944 |    1099712 | Rdr |50  00  57  cd                                                           |     | 
    2260240 |    2261232 | Rdr |52                                                                       |     | 
    2347424 |    2349888 | Rdr |93  20                                                                   |     | 
    2351076 |    2356964 | Tag |a4  e0  75  6a  5b                                                       |     | 
    2460448 |    2470976 | Rdr |93  70  a4  e0  75  6a  5b  d4  67                                       |     | 
    2628560 |    2637936 | Rdr |cf  ef  be  ad  de  c6  34  15                                
```

And it should look like:

```
      Start |        End | Src | Data (! denotes parity error)                                           | CRC | Annotation
------------+------------+-----+-------------------------------------------------------------------------+-----+--------------------
          0 |        992 | Rdr |52                                                                       |     | 
       2116 |       4484 | Tag |04  00                                                                   |     | 
       7040 |       9504 | Rdr |93  20                                                                   |     | 
      10564 |      16452 | Tag |a4  e0  75  6a  5b                                                       |     | 
      18944 |      29472 | Rdr |93  70  a4  e0  75  6a  5b  d4  67                                       |     | 
      30532 |      34052 | Tag |08  b6  dd                                                               |     | 
      44288 |      53664 | Rdr |cf  de  ad  be  ef  c6  46  e9                                           |     | 
      57924 |      97220 | Tag |00  00  de  ad  be  ef  02  00  09  78  00  91  02  da  bc  19  10  10   |     | 
            |            |     |11  12  13  14  15  16  04  00  08  00  89  f8  a3  5c  69  0b
```

As you can see from the reader log, the password is sent in the LSB order and it should be MSB order instead?

### Reproduction

Write a Gen4 card with the config shown above;
Try to authenticate using DEADBEEF as the password.

sfjuocekr, 2023-07-30:
I have to add I noticed this behavior when trying to figure out why writing a card to my Gen4 would put it in a non-working state.

The weird thing is, I couldn't recover it using the PM3 but when I did a wipe with FZ I was able to detect the card again on the PM3 and set it up again with the PM3.

I might also have had CL2/3 skip on by accident last night when I was mocking with it, I was pretty tired when I stumbled over this.

skotopes, 2023-08-01:
Looks like there is a whole zoo of gen4. We are researching this topic but it's quite difficult to do that without real cards.

Astrrra, 2023-08-02:
Hi! Can you please check what is the output of the `CF00000000CC` command?

UrSuLa360, 2023-08-22:
Try to invert the endian of the password.

From DEADBEEF to EFBEADDE

doomwastaken, 2023-11-03:
@Astrrra need your input on what do we do with this

skotopes, 2023-11-08:
@sfjuocekr @UrSuLa360 

It looks like this issue was fixed in latest Magic app update. Please retest and reopen issue if problem persist.

```
[usb] pm3 --> hf 14a raw -s -c -t 1000 cf00000000CC
[+] 00 00 00 06 A0 [ 8C F1 ]
[usb] pm3 --> hf 14a s
sim    sniff  
[usb] pm3 --> hf 14a sniff 

[#] Starting to sniff. Press PM3 Button to stop.
[#] trace len = 365
[usb] pm3 --> hf 14a list 
[=] downloading tracelog data from device
[+] Recorded activity (trace len = 365 bytes)
[=] start = start of start frame end = end of frame. src = source of transfer
[=] ISO14443A - all times are in carrier periods (1/13.56MHz)

      Start |        End | Src | Data (! denotes parity error)                                           | CRC | Annotation
------------+------------+-----+-------------------------------------------------------------------------+-----+--------------------
          0 |        992 | Rdr |40(7)                                                                    |     | MAGIC WUPC1
     464640 |     465632 | Rdr |52(7)                                                                    |     | WUPA
     466884 |     469252 | Tag |44  00                                                                   |     | 
     473824 |     476288 | Rdr |93  20                                                                   |     | ANTICOLL
     477476 |     483300 | Tag |88  04  82  46  48                                                       |     | 
     488032 |     498496 | Rdr |93  70  88  04  82  46  48  ab  1e                                       |  ok | SELECT_UID
     499748 |     503268 | Tag |04  da  17                                                               |  ok | 
     508080 |     510544 | Rdr |95  20                                                                   |     | ANTICOLL-2
     511732 |     517620 | Tag |4a  9d  69  80  3e                                                       |     | 
     522224 |     532752 | Rdr |95  70  4a  9d  69  80  3e  f9  a4                                       |  ok | SELECT_UID-2
     533940 |     537460 | Tag |08  b6  dd                                                               |  ok | 
     542128 |     551504 | Rdr |cf  00  00  00  00  c6  73  ae                                           |  ok | 
     555892 |     583924 | Tag |00  01  00  00  00  00  02  00  00  00  00  00  00  00  00  00  00  00   |     | 
            |            |     |00  00  00  00  00  00  00                                               |  !! | 
  504639008 |  504640000 | Rdr |52(7)                                                                    |     | WUPA
  504641236 |  504643604 | Tag |44  00                                                                   |     | 
  504648064 |  504650528 | Rdr |93  20                                                                   |     | ANTICOLL
  504651700 |  504657524 | Tag |88  04  82  46  48                                                       |     | 
  504662096 |  504672560 | Rdr |93  70  88  04  82  46  48  ab  1e                                       |  ok | SELECT_UID
  504673796 |  504677316 | Tag |04  da  17                                                               |  ok | 
  504681792 |  504684256 | Rdr |95  20                                                                   |     | ANTICOLL-2
  504685428 |  504691316 | Tag |4a  9d  69  80  3e                                                       |     | 
  504695872 |  504706400 | Rdr |95  70  4a  9d  69  80  3e  f9  a4                                       |  ok | SELECT_UID-2
  504707572 |  504711092 | Tag |08  b6  dd                                                               |  ok | 
  504715696 |  504729680 | Rdr |cf  00  00  00  00  fe  01  23  45  67  36  31                           |  ok | 
  504807412 |  504812148 | Tag |90  00  fd  07                                                           |  ok | 
[usb] pm3 --> hf 14a raw -s -c -t 1000 cf00000000CC
[usb] pm3 --> hf 14a raw -s -c -t 1000 cf01234567CC
[+] 00 00 00 06 A0 [ 8C F1 ]
[usb] pm3 --> hf 14a list
[=] downloading tracelog data from device
[+] Recorded activity (trace len = 166 bytes)
[=] start = start of start frame end = end of frame. src = source of transfer
[=] ISO14443A - all times are in carrier periods (1/13.56MHz)

      Start |        End | Src | Data (! denotes parity error)                                           | CRC | Annotation
------------+------------+-----+-------------------------------------------------------------------------+-----+--------------------
          0 |        992 | Rdr |52(7)                                                                    |     | WUPA
       2100 |       4468 | Tag |44  00                                                                   |     | 
       7040 |       9504 | Rdr |93  20                                                                   |     | ANTICOLL
      10548 |      16372 | Tag |88  04  82  46  48                                                       |     | 
      19072 |      29536 | Rdr |93  70  88  04  82  46  48  ab  1e                                       |  ok | SELECT_UID
      30644 |      34164 | Tag |04  da  17                                                               |  ok | 
      35584 |      38048 | Rdr |95  20                                                                   |     | ANTICOLL-2
      39092 |      44980 | Tag |4a  9d  69  80  3e                                                       |     | 
      47616 |      58144 | Rdr |95  70  4a  9d  69  80  3e  f9  a4                                       |  ok | SELECT_UID-2
      59188 |      62708 | Tag |08  b6  dd                                                               |  ok | 
      73728 |      83040 | Rdr |cf  01  23  45  67  cc  65  b7                                           |  ok | 
      86580 |      94772 | Tag |00  00  00  06  a0  8c  f1                                               |  ok | 
```

UrSuLa360, 2023-11-09:
I cannot test, i don't have a gen4  card anymore.
If anyone can test it is super fast.
Lock your **Gen4 card** with a password like 12345678 (avoid FFFFFFFF and similars) with your favourite tool as iCopy-XS, Proxmark3, LibNFC or similar.

And try to unlock and read the card with flipper using **NFC Magic** app latest version. It will requiere the password you entered before. 12345678 should unlocks it, if fails try 78563412.
write here what happened with the test. Thanks

skotopes, 2023-11-09:
@UrSuLa360 there is a fix coming after 0.95 release

skotopes, 2024-01-10:
@sfjuocekr @UrSuLa360 please check latest version. Should be fixed. Reopen if not

closed 2024-01-10 by skotopes