flipperdevices/flipperzero-firmware #2930: Support for Rubber Ducky Scripting Language Version 3 in Flipper Zero

solomyr, 2023-07-31:
### Description of the feature you're suggesting.

As an avid user and enthusiast of the Flipper Zero device, I have found it to be an incredibly versatile and powerful tool. However, there is one feature that I believe could significantly enhance its functionality and usability: the support for Rubber Ducky Scripting Language version 3.

Rubber Ducky Script Language (RDScript) is a powerful scripting language used for automating keystrokes on a target device, making it a valuable tool for penetration testers and cybersecurity enthusiasts. The latest version, RDScript V3, has introduced several new features and improvements that make it even more potent and flexible.

By adding support for RDScript V3, Flipper Zero could:

Enable users to write more complex and powerful scripts.
Increase the device's versatility and appeal to a broader user base.
Keep up with the latest advancements in the field of cybersecurity.
I believe that this feature would be a valuable addition to the Flipper Zero's capabilities and would greatly appreciate it if the development team could consider implementing it in a future update.

Thank you for your time and consideration. I look forward to seeing how Flipper Zero continues to evolve and improve.

skotopes, 2023-08-01:
RDScript V3 is a proprietary technology. We don't plan to support it.

closed 2023-08-01 by skotopes

melroy89, 2023-09-30:
Reading this.. I'm actually more willing to buy a flipper device rather than giving money to hak5.

hackerdolphin, 2025-07-20:
ok it'S fine