flipperdevices/flipperzero-firmware #2950: Uploading firmware to Flipper stops randomly

krupers, 2023-08-04:
### Describe the bug.

With latest releases update 0.88.0 I am encountering an issue with update. I am using a mobile app and usually it takes 5-10 minutes to update the firmware. With 0.88.0 I am trying to update and it pauses on different moments of percentage progress. I tried to reboot the Flipper OS, the app, abandon Sync, and it always stops at random progress. 

### Reproduction

1. Reboot the Flipper OS
2. Start the Flipper app (iOS)
3. Wait for Flipper to be connected and synced
4. Click on update button to upgrade from 0.87.0 to 0.88.0

### Target

Flipper zero update

### Anything else?

Tried this multiple times and maximum progress bar reached was 43%

krupers, 2023-08-05:
Okay, I have found a solution, I think. I have noticed that after previous update my PIN disappeared and I was forced to create new one again. Then I charged flipper battery and then everything was okay. Now I have latest firmware. But this is strange.

doomwastaken, 2023-08-05:
Could you provide version of iOS app, so we could try and reproduce it? We've tested with latest iOS app during release

krupers, 2023-08-05:
Sure, my phone is iPhone 14 Pro Max (iOS 16.6) with installed latest Flipper App (v1.6.4).
Always when I am having an old App it automatically notices me about the update.

doomwastaken, 2023-08-05:
Alright, thank you for details. We will try to experiment   with locking the device

skotopes, 2023-08-05:
Next time you see this issue please try changing Settings->System->Sleep Method to legacy

krupers, 2023-08-05:
Hey, thank you for the tip. Unfortunately I already have this mode enabled, because of that Core error we all having last time :( So this mode is not helping. Maybe should I disable it then?

skotopes, 2023-08-08:
@krupers some logs from the device will help

krupers, 2023-08-08:
All the logs from the moment I have tried to update were attached to the first post. I hope they will help

skotopes, 2023-08-08:
Logs says that data was transferred till the last moment before disconnect.
Let's move this issue to appropriate github repo for further debug.