flipperdevices/flipperzero-firmware #2952: Issue with path containing spaces

tompecina, 2023-08-05:
### Describe the bug.

I found that the howto/tutorial app `https://github.com/csBlueChip/FlipperZero_plugin_howto/tree/main` could not be launched. The reason was a missing pair of quotes in the command sent to the storage. I fixed it in the source code as follows:

```
[tompecina@desktop flipperZero]$ diff -u official/scripts/runfap.py mod/scripts/runfap.py 
--- official/scripts/runfap.py	2023-08-04 19:59:14.958412832 +0200
+++ mod/scripts/runfap.py	2023-08-05 12:47:15.788945957 +0200
@@ -68,7 +68,7 @@
                     startup_command = self.args.host_app
 
                 self.logger.info(f"Launching app: {startup_command}")
-                storage.send_and_wait_eol(f"loader open {startup_command}\r")
+                storage.send_and_wait_eol(f"loader open \"{startup_command}\"\r")
 
                 if len(result := storage.read.until(storage.CLI_EOL)):
                     self.logger.error(f"Unexpected response: {result.decode('ascii')}")

```

### Reproduction

1. Follow the instructions in the howto.
2. Launch the app by issuing **./fbt launch APPSRC=bc_demo**
3. The error is reported and the app does not start.

doomwastaken, 2023-08-05:
https://github.com/flipperdevices/flipperzero-firmware/issues/2947

doomwastaken, 2023-08-05:
Duplicated

closed 2023-08-05 by skotopes