flipperdevices/flipperzero-firmware #2979: RAW SubGHz Signal Editor

AmongusGaming, 2023-08-11:
### Description of the feature you're suggesting.

An editor for RAW SubGHz files where you can cut off certain parts of a signal to get a cleaner signal. I know you can get a good signal with RSSI Threshold, but it would be useful to be able to edit the RAW capture later on. in addition, you should be able to merge raw signals together, so you can have one file to trigger multiple signals multiple times. this could be useful if you’re trying to combine a raw signal from the internet and one raw signal that you have recorded.

skotopes, 2023-08-23:
We don't store RSSI value in RAW file, that make impossible to make meaningful editor for RAW files.

closed 2023-08-23 by skotopes

skotopes, 2023-08-24:
@AmongusGaming you do understand what exactly we store in subghz RAW files? If you do understand, can you imagine some editing tool for it? How it will look like? 

Anyway, there are no plans on designing such tools from our side, but we may revisit this topic later.

AmongusGaming, 2023-08-24:
> @AmongusGaming you do understand what exactly we store in subghz RAW files? If you do understand, can you imagine some editing tool for it? How it will look like? 
> 
> 
> 
> Anyway, there are no plans on designing such tools from our side, but we may revisit this topic later.

Ok, fair enough.

Lechnio, 2026-06-07:
> > @AmongusGaming you do understand what exactly we store in subghz RAW files? If you do understand, can you imagine some editing tool for it? How it will look like?
> > Anyway, there are no plans on designing such tools from our side, but we may revisit this topic later.
> 
> Looking back on this thread I know realize how stupid I sound in my response, haha

Everyone's learning all the time :D For me this project is a good way to understand how signal processing works in Flipper ecosystem. 
For example today I've just added feature to load KeeLoq recognized signals as RAW again. So you can use regular Read and then save it all as RAWs. What I learned is 'decode' feature requires some additional empty data (~1.5ms) to properly decode RAW as KeeLoq again. I was to lazy to go out and check if the garage gate will understand the signal without this padding haha It would usefull gap between seperate merged signals tho