flipperdevices/flipperzero-firmware #2981: Unknown RFID (125 kHz) protocol identification/support request

FerralFred, 2023-08-14:
### Description of the feature you're suggesting.

We recently got new cards to access the trash room in our building. It's a generic white card with no text on it. Flipper identifies the reader as 125kHz. I created a raw dump. ASK is empty, PSK is here: [RfidRecord22.psk.zip](https://github.com/flipperdevices/flipperzero-firmware/files/12334739/RfidRecord22.psk.zip)

Can anyone identify which type of card that is and if there is any chance flipper will support it?

skotopes, 2023-08-25:
@FerralFred this recording doesn't contains any meaningful data, most likely this card is NFC or Reader Talk First RFID. Anyway we need more details on reader and system.

skotopes, 2024-01-10:
We need system/card vendor and model number. Please reopen when you get it.

closed 2024-01-10 by skotopes