flipperdevices/flipperzero-firmware #2983: Flipper crashes when opening a "Cannot load key file" file by pressing down.

ngonzalez456, 2023-08-16:
### Describe the bug.

Flipper crashes on firmware v.0.88.0 US using a 32GB SD card.

Precondition: Having a RFID defective key that throws the "Cannot load key file" when selected.

### Reproduction

1. Go to home screen (Flipper animation screen).
2. Press down.
3. Select from Favorites or press right to find the defective file from the precondition.
4. Verify the "Cannot load key file" screen is shown.
5. Press the Back button.
ERR1: Flipper crashes. BusFault error when recovered.

### Target

Develop

### Logs

```Text
18 [default] Binding on background is not deferred as requested by the DeferredPropertyNames class info because one or more of its sub-objects contain an id.
18 [default] Binding on contentItem is not deferred as requested by the DeferredPropertyNames class info because one or more of its sub-objects contain an id.
19 [default] Binding on contentItem is not deferred as requested by the DeferredPropertyNames class info because one or more of its sub-objects contain an id.
19 [default] Binding on background is not deferred as requested by the DeferredPropertyNames class info because one or more of its sub-objects contain an id.
20 [default] Binding on background is not deferred as requested by the DeferredPropertyNames class info because one or more of its sub-objects contain an id.
20 [default] Binding on contentItem is not deferred as requested by the DeferredPropertyNames class info because one or more of its sub-objects contain an id.
20 [default] Binding on animation is not deferred as requested by the DeferredPropertyNames class info because one or more of its sub-objects contain an id.
23 [default] Binding on background is not deferred as requested by the DeferredPropertyNames class info because one or more of its sub-objects contain an id.
24 [default] Binding on background is not deferred as requested by the DeferredPropertyNames class info because one or more of its sub-objects contain an id.
24 [default] Binding on background is not deferred as requested by the DeferredPropertyNames class info because one or more of its sub-objects contain an id.
24 [default] Binding on contentItem is not deferred as requested by the DeferredPropertyNames class info because one or more of its sub-objects contain an id.
24 [default] Binding on indicator is not deferred as requested by the DeferredPropertyNames class info because one or more of its sub-objects contain an id.
122 [APP] qFlipper version 1.3.2 commit 7b0839e1 2023-06-08T04:58:22
122 [APP] OS info: macOS 13.4 13.4 22.5.0 Qt 6.3.1
140 [REG] Detected new device: VID_0x483:PID_0x5740
182 [DBG] Trying serial port  at /dev/cu.wlan-debug
182 [DBG] Trying serial port  at /dev/tty.wlan-debug
182 [DBG] Trying serial port  at /dev/cu.Bluetooth-Incoming-Port
182 [DBG] Trying serial port  at /dev/tty.Bluetooth-Incoming-Port
182 [DBG] Trying serial port flip_Amkavota at /dev/cu.usbmodemflip_Amkavota1
182 [DBG] Using  serial port flip_Amkavota at /dev/cu.usbmodemflip_Amkavota1
189 [RPC] Starting RPC session...
308 [RPC] RPC session started successfully.
308 [RPC] (1) System Protobuf Version START
360 [UPD] Fetched update information from https://update.flipperzero.one/qFlipper/directory.json
366 [DBG] Detected protobuf version: 0.18
366 [RPC] (1) System Protobuf Version SUCCESS
366 [RPC] (2) Property Get START
430 [UPD] Fetched update information from https://update.flipperzero.one/firmware/directory.json
510 [RPC] (2) Property Get SUCCESS
511 [RPC] (3) Storage Info @/ext START
562 [RPC] (3) Storage Info @/ext SUCCESS
563 [RPC] (4) Storage Stat @/ext/Manifest START
616 [RPC] (4) Storage Stat @/ext/Manifest SUCCESS
616 [RPC] (5) System Get DateTime START
668 [DBG] Flipper time skew is -675 milliseconds
668 [RPC] (5) System Get DateTime SUCCESS
668 [RPC] (6) System Set DateTime START
720 [RPC] (6) System Set DateTime SUCCESS
721 [RPC] Stopping RPC session...
731 [RPC] RPC session stopped successfully.
732 [REG] Registering the device
732 [DEV] Version: 0.88.0 commit: 357fbfe6 radio: 1.15.0
732 [BKD] Current device changed to "Amkavota"
834 [RPC] Starting RPC session...
957 [RPC] RPC session started successfully.
958 [RPC] (1) Gui Start ScreenStream START
1015 [RPC] (1) Gui Start ScreenStream SUCCESS
8397 [RPC] Serial connection was lost.
8397 [RPC] Stopping RPC session...
8398 [RPC] RPC session stopped successfully.
8444 [REG] Device disconnected: VID_0x483:PID_0x5740
8444 [BKD] Last device was disconnected
9256 [REG] Detected new device: VID_0x483:PID_0x5740
9258 [DBG] Trying serial port  at /dev/cu.wlan-debug
9258 [DBG] Trying serial port  at /dev/tty.wlan-debug
9258 [DBG] Trying serial port  at /dev/cu.Bluetooth-Incoming-Port
9258 [DBG] Trying serial port  at /dev/tty.Bluetooth-Incoming-Port
9258 [DBG] Trying serial port flip_Amkavota at /dev/cu.usbmodemflip_Amkavota1
9258 [DBG] Using  serial port flip_Amkavota at /dev/cu.usbmodemflip_Amkavota1
9258 [RPC] Starting RPC session...
9395 [RPC] RPC session started successfully.
9395 [RPC] (1) System Protobuf Version START
10505 [DBG] Detected protobuf version: 0.18
10505 [RPC] (1) System Protobuf Version SUCCESS
10505 [RPC] (2) Property Get START
10655 [RPC] (2) Property Get SUCCESS
10655 [RPC] (3) Storage Info @/ext START
10708 [RPC] (3) Storage Info @/ext SUCCESS
10709 [RPC] (4) Storage Stat @/ext/Manifest START
10762 [RPC] (4) Storage Stat @/ext/Manifest SUCCESS
10762 [RPC] (5) System Get DateTime START
10814 [DBG] Flipper time skew is -1821 milliseconds
10814 [RPC] (5) System Get DateTime SUCCESS
10815 [RPC] (6) System Set DateTime START
10867 [RPC] (6) System Set DateTime SUCCESS
10867 [RPC] Stopping RPC session...
10877 [RPC] RPC session stopped successfully.
10878 [REG] Registering the device
10878 [DEV] Version: 0.88.0 commit: 357fbfe6 radio: 1.15.0
10878 [BKD] Current device changed to "Amkavota"
10979 [RPC] Starting RPC session...
11096 [RPC] RPC session started successfully.
11097 [RPC] (1) Gui Start ScreenStream START
11155 [RPC] (1) Gui Start ScreenStream SUCCESS
```

### Anything else?

I can share a video if needed through a link.

hedger, 2023-08-18:
Please share the file causing the issue. Where did you get it from?

ngonzalez456, 2023-08-18:
[Gimnasio_98.rfid.zip](https://github.com/flipperdevices/flipperzero-firmware/files/12374252/Gimnasio_98.rfid.zip)

There's the file. I got it by scanning my gym's card. Nothing fancy.

PR #3012 (2023-08-25): [FL-3565] Rfid: fix crash on broken key launch from archive
# What's new

- Rfid: fix crash on broken key launch from archive, fixes #2983

# Verification 

- Open broken rfid key from archive (#2983)

# Checklist (For Reviewer)

- [x] PR has description of feature/bug or link to Confluence/Jira task
- [x] Description contains actions to verify feature/bugfix
- [ ] I've built this code, uploaded it to the device and verified feature/bugfix
closed 2023-08-25 by hedger