flipperdevices/flipperzero-firmware #2992: Toolchain version not supported

pengu1ndev, 2023-08-17:
### Describe the bug.

I cloned the dev branch using:
`git clone --recursive https://github.com/flipperdevices/flipperzero-firmware.git`

and tried to build it with `./fbt` but after downloading the toolchain it gave me the error:

scons: *** Toolchain version is not supported. Allowed: (' 10.3.',), toolchain:    Stop.

The error occurs on both my macos and windows machine and compiling the release branch seems to work fine.

### Reproduction

1. Clone dev branch
2. build using ./fbt
3. toolchain version not supported error

### Anything else?

I tried some solutions with some help from the official discord server:

- Using `./fbt -c` to clean the files  which sends the same toolchain error
- `git checkout dev`
`git clean -dxf`
`git submodule update --init --recursive --remote` and then I tried to compile it using `./fbt` again and I got the same error.

hedger, 2023-08-18:
Could you please remove the 'toolchain' folder from firmware's folder on problematic machines and try again? Please post full logs of first 'fbt' Run after that, including toolchain download.

pengu1ndev, 2023-08-18:
> Could you please remove the 'toolchain' folder from firmware's folder on problematic machines and try again? Please post full logs of first 'fbt' run after that, including toolchain download.

./fbt
Checking for tar..yes
Checking if downloaded toolchain tgz exists..no
Checking curl..yes
Downloading toolchain:
################################################################################################################### 100.0%
done
Removing old toolchain..done
Unpacking toolchain to '/Users/bendf/Documents/Flipper Development/dev/flipperzero-firmware/toolchain':
##################################################################### 100.0%
done
Cleaning up..done
scons: *** Toolchain version is not supported. Allowed: (' 10.3.',), toolchain:    Stop.

hedger, 2023-08-18:
Could you please run `source scripts/toolchain/fbtenv.sh` and show the output of `which arm-none-eabi-gcc`, `arm-none-eabi-gcc` and `env` or `printenv`?

pengu1ndev, 2023-08-18:
> Could you please run `source scripts/toolchain/fbtenv.sh` and show the output of `which arm-none-eabi-gcc`, `arm-none-eabi-gcc` and `env` or `printenv`?

`which arm-none-eabi-gcc`:
/Users/bendf/Documents/Flipper Development/dev/flipperzero-firmware/toolchain/x86_64-darwin/bin/arm-none-eabi-gcc

`arm-none-eabi-gcc`:
arm-none-eabi-gcc: fatal error: no input files
compilation terminated.

`env`:
COLORTERM=truecolor
COMMAND_MODE=unix2003
HOME=/Users/bendf
LANG=en_GB.UTF-8
LOGNAME=bendf
PATH=/Users/bendf/Documents/Flipper Development/dev/flipperzero-firmware/toolchain/x86_64-darwin/openssl/bin:/Users/bendf/Documents/Flipper Development/dev/flipperzero-firmware/toolchain/x86_64-darwin/openocd/bin:/Users/bendf/Documents/Flipper Development/dev/flipperzero-firmware/toolchain/x86_64-darwin/protobuf/bin:/Users/bendf/Documents/Flipper Development/dev/flipperzero-firmware/toolchain/x86_64-darwin/bin:/Users/bendf/Documents/Flipper Development/dev/flipperzero-firmware/toolchain/x86_64-darwin/python/bin:/Users/bendf/.pyenv/shims:/usr/local/bin:/System/Cryptexes/App/usr/bin:/usr/bin:/bin:/usr/sbin:/sbin:/Applications/VMware Fusion.app/Contents/Public:/Library/Apple/usr/bin:/Applications/Wireshark.app/Contents/MacOS:/var/run/com.apple.security.cryptexd/codex.system/bootstrap/usr/local/bin:/var/run/com.apple.security.cryptexd/codex.system/bootstrap/usr/bin:/var/run/com.apple.security.cryptexd/codex.system/bootstrap/usr/appleinternal/bin
SHELL=/bin/zsh
SSH_AUTH_SOCK=/private/tmp/com.apple.launchd.O6Eu5aBB9k/Listeners
SSH_SOCKET_DIR=~/.ssh
TERM=xterm-256color
TERM_PROGRAM=WarpTerminal
TERM_PROGRAM_VERSION=v0.2023.08.15.08.03.stable_00
TMPDIR=/var/folders/sx/9t53t84968v91p974ys3x2280000gn/T/
USER=bendf
WARP_IS_LOCAL_SHELL_SESSION=1
WARP_USE_SSH_WRAPPER=1
XPC_FLAGS=0x0
XPC_SERVICE_NAME=0
__CFBundleIdentifier=dev.warp.Warp-Stable
__CF_USER_TEXT_ENCODING=0x1F5:0:2
SHLVL=1
PWD=/Users/bendf/Documents/Flipper Development/dev/flipperzero-firmware
OLDPWD=/Users/bendf/Documents/Flipper Development/dev
PYENV_SHELL=zsh
CONDA_CHANGEPS1=false
SAVED_SSL_CERT_FILE=
SAVED_REQUESTS_CA_BUNDLE=
SAVED_PYTHONNOUSERSITE=
SAVED_PYTHONPATH=
SAVED_PYTHONHOME=
SSL_CERT_FILE=/Users/bendf/Documents/Flipper Development/dev/flipperzero-firmware/toolchain/x86_64-darwin/python/lib/python3.11/site-packages/certifi/cacert.pem
REQUESTS_CA_BUNDLE=/Users/bendf/Documents/Flipper Development/dev/flipperzero-firmware/toolchain/x86_64-darwin/python/lib/python3.11/site-packages/certifi/cacert.pem
PYTHONNOUSERSITE=1
PYTHONPATH=
PYTHONHOME=
_=/usr/bin/env

`printenv`:
COLORTERM=truecolor
COMMAND_MODE=unix2003
HOME=/Users/bendf
LANG=en_GB.UTF-8
LOGNAME=bendf
PATH=/Users/bendf/Documents/Flipper Development/dev/flipperzero-firmware/toolchain/x86_64-darwin/openssl/bin:/Users/bendf/Documents/Flipper Development/dev/flipperzero-firmware/toolchain/x86_64-darwin/openocd/bin:/Users/bendf/Documents/Flipper Development/dev/flipperzero-firmware/toolchain/x86_64-darwin/protobuf/bin:/Users/bendf/Documents/Flipper Development/dev/flipperzero-firmware/toolchain/x86_64-darwin/bin:/Users/bendf/Documents/Flipper Development/dev/flipperzero-firmware/toolchain/x86_64-darwin/python/bin:/Users/bendf/.pyenv/shims:/usr/local/bin:/System/Cryptexes/App/usr/bin:/usr/bin:/bin:/usr/sbin:/sbin:/Applications/VMware Fusion.app/Contents/Public:/Library/Apple/usr/bin:/Applications/Wireshark.app/Contents/MacOS:/var/run/com.apple.security.cryptexd/codex.system/bootstrap/usr/local/bin:/var/run/com.apple.security.cryptexd/codex.system/bootstrap/usr/bin:/var/run/com.apple.security.cryptexd/codex.system/bootstrap/usr/appleinternal/bin
SHELL=/bin/zsh
SSH_AUTH_SOCK=/private/tmp/com.apple.launchd.O6Eu5aBB9k/Listeners
SSH_SOCKET_DIR=~/.ssh
TERM=xterm-256color
TERM_PROGRAM=WarpTerminal
TERM_PROGRAM_VERSION=v0.2023.08.15.08.03.stable_00
TMPDIR=/var/folders/sx/9t53t84968v91p974ys3x2280000gn/T/
USER=bendf
WARP_IS_LOCAL_SHELL_SESSION=1
WARP_USE_SSH_WRAPPER=1
XPC_FLAGS=0x0
XPC_SERVICE_NAME=0
__CFBundleIdentifier=dev.warp.Warp-Stable
__CF_USER_TEXT_ENCODING=0x1F5:0:2
SHLVL=1
PWD=/Users/bendf/Documents/Flipper Development/dev/flipperzero-firmware
OLDPWD=/Users/bendf/Documents/Flipper Development/dev
PYENV_SHELL=zsh
CONDA_CHANGEPS1=false
SAVED_SSL_CERT_FILE=
SAVED_REQUESTS_CA_BUNDLE=
SAVED_PYTHONNOUSERSITE=
SAVED_PYTHONPATH=
SAVED_PYTHONHOME=
SSL_CERT_FILE=/Users/bendf/Documents/Flipper Development/dev/flipperzero-firmware/toolchain/x86_64-darwin/python/lib/python3.11/site-packages/certifi/cacert.pem
REQUESTS_CA_BUNDLE=/Users/bendf/Documents/Flipper Development/dev/flipperzero-firmware/toolchain/x86_64-darwin/python/lib/python3.11/site-packages/certifi/cacert.pem
PYTHONNOUSERSITE=1
PYTHONPATH=
PYTHONHOME=
_=/usr/bin/printenv

pengu1ndev, 2023-08-20:
Massive error on my side, I had multiple versions of the firmware cloned into the same folder so I believe they were just conflicting. Thanks for any help.

closed 2023-08-20 by pengu1ndev

SpeedyGo55, 2023-09-08:
how did you solve it? I have the same problem but i only have one version of the fw in the folder.

pengu1ndev, 2023-09-08:
> how did you solve it? I have the same problem but i only have one version of the fw in the folder.

It happens when you have a folder with spaces in it in the path to your firmware folder