flipperdevices/flipperzero-firmware #2995: RFID tag not readable

Tylerrattv, 2023-08-21:
### Describe the bug.

Hi, 

Got a unreadable tag. I believe this a "Farpointe Pyramid MAXSecure" format on a T5577 tag. 
Have attached raw RFID files.
[Unreadable.zip](https://github.com/flipperdevices/flipperzero-firmware/files/12390646/Unreadable.zip)

### Reproduction

RFID Read

Tylerrattv, 2023-10-22:
Is there any update to this tag being supported?

skotopes, 2023-10-30:
Nope, it's in backlog

crunchy234, 2023-12-05:
Hopefully this helps but this has been implemented on the Keysy RFID duplicator: https://medium.com/@ninjavelli/anytime-fitness-hacked-rfid-nfc-tag-duplicator-vs-access-control-systems-9e9d175766e7 

Very keen to have this feature working. Also can test dev builds of this if that is helpful.

crunchy234, 2023-12-06:
Just looking at the firmware code, it seems like this is supported: https://github.com/flipperdevices/flipperzero-firmware/blob/dev/lib/lfrfid/protocols/protocol_pyramid.c 

I can read my card successfully but can't emulate it successfully.

During an emulation, the reader beeps as if it's a real card, but then the door doesn't open 🤔

crunchy234, 2023-12-10:
I have been using this release https://github.com/Flipper-XFW/Xtreme-Firmware/releases/tag/XFW-0052_09122023 
Which makes emulation work on the door aka Beep & unlock door...

I'm using the same scan from v0.95 - stable standard firmware

IMLJ2009, 2023-12-18:
I have the same issues, do you successfully solve this problem? thanks.

skotopes, 2024-01-10:
@gornekich @doomwastaken any progress?

Ak-d0V, 2024-12-27:
> Any update?

@bwt615 Farpointe Pyramid is working on my Flipper. version unlshd-079
which error are you getting ?

bwt615, 2024-12-27:
No detection of AF key fob on 1.1.2 :(

alzzzy, 2026-11-03:
update on this?

Ak-d0V, 2026-11-03:
> 19\. update on this?

Pyramid works natively; what problem are you encountering ?

alzzzy, 2026-11-09:
> > 19. update on this?
> 
> Pyramid works natively; what problem are you encountering ?

i cant still clone my af keyfob. Can you share how you do it?

Ak-d0V, 2026-11-09:
Upload a picture of your the keyfob you are trying to copy. And on your flipper, go to both IRID and NFC and try to scan the tag. What ever works, post a picture of the reading (you can edit the picture and black out sensitive info)

alzzzy, 2026-11-10:
I try both irid and nfc not working
[image]

Ak-d0V, 2026-11-15:
lol that's Anytime Fitness security key. you can't copy it using your flipper. even if the protocol is based on pyramid it also has maxsecure. so yeah it's neither RFID nor NFC frequencies. not going to work with basic cloning equipment.