flipperdevices/flipperzero-firmware #3003: System Crash on furi_hal_bus_disable during Nero Radio 56bit Key Emulation in SubHz

GLinnik21, 2023-08-22:
### Describe the bug.

When attempting to retransmit a key using Nero Radio 56bit, the system crashes during the "Send" operation in the Emulate mode.

Crashlog:
```
__furi_crash@0x0800f8e0 (furi/core/check.c:166)
furi_hal_bus_disable@0x080029d8 (firmware/targets/f7/furi_hal/furi_hal_bus.c:269)
furi_hal_subghz_stop_async_tx@0x0800c222 (firmware/targets/f7/furi_hal/furi_hal_subghz.c:739)
subghz_txrx_tx_stop@0x2000cd7c (applications/main/subghz/helpers/subghz_txrx.c:325)
subghz_txrx_stop@0x2000cf92 (applications/main/subghz/helpers/subghz_txrx.c:355)
subghz_scene_transmitter_on_event@0x20010df2 (applications/main/subghz/scenes/subghz_scene_transmitter.c:71)
scene_manager_handle_custom_event@0x0807c4d6 (applications/services/gui/scene_manager.c:56)
subghz_custom_event_callback@0x200134f2 (applications/main/subghz/subghz.c:8)
view_dispatcher_handle_custom_event@0x0807d098 (applications/services/gui/view_dispatcher.c:307)
view_dispatcher_run@0x0807d4f4 (applications/services/gui/view_dispatcher.c:93)
subghz_app@0x200139a4 (applications/main/subghz/subghz.c:295)
flipper_application_thread@0x0805c204 (lib/flipper_application/flipper_application.c:219)
furi_thread_body@0x08015290 (furi/core/thread.c:90)
furi_thread_catch@0x080151ca (furi/core/thread.c:58)
```

### Reproduction

1. Navigate to SubHz.
1. Go to Saved.
1. Select [My key].
1. Choose Emulate.
1. Click on Send.
1. The system crashes.

### Target

0.89.0 built with DEBUG env 

### Logs

```Text
0 [I][FuriHalRtc] Init OK
2 [I][FuriHalInterrupt] Init OK
4 [E][FuriHalFlash] FLASH->SR 0x00008000
6 [I][FuriHalResources] Init OK
8 [I][FuriHalVersion] Init OK
10 [I][FuriHalSpiConfig] Init OK
12 [I][FuriHalIbutton] Init OK
13 [I][FuriHalSpeaker] Init OK
14 [I][FuriHalCrypto] Init OK
15 [I][FuriHalI2C] Init OK
19 [I][Gauge] Device Number 0220
22 [I][Gauge] Verifying data memory
82 [I][Gauge] Data memory verification complete
86 [I][FuriHalPower] Init OK
90 [I][FuriHalMemory] SRAM2A: 0x20031378, 13448
93 [I][FuriHalMemory] SRAM2B: 0x20038000, 11264
95 [I][FuriHalMemory] SRAM2A clear
98 [I][Core2] Core2 started
101 [I][FuriHalMemory] SRAM2B clear
104 [I][FuriHalMemory] Enabled
106 [I][FuriHalUsb] Init OK
108 [I][FuriHalVibro] Init OK
110 [I][FuriHalSubGhz] Init OK
235 [I][FuriHalNfc] Init OK
236 [I][Flipper] 
	Firmware version:	0.89.0-path
	Build date:		23-08-2023
	Git Commit:		9ede97d7 (0) (dirty)
	Git Branch:		no-restrictions
239 [I][Flipper] Boot mode 0, starting services
241 [D][Flipper] Starting service BtSrv
243 [D][Flipper] Starting service CliSrv
246 [I][SavedStruct] Loading "/int/.bt.settings"
250 [D][Flipper] Starting service DialogsSrv
254 [D][Flipper] Starting service DolphinSrv
258 [I][CliVcp] Init OK
260 [I][FuriHalUsb] USB Mode change done
263 [I][SavedStruct] Loading "/int/.dolphin.state"
266 [D][Flipper] Starting service DesktopSrv
269 [D][CliVcp] Start
271 [D][Flipper] Starting service GuiSrv
274 [D][Flipper] Starting service InputSrv
277 [D][Flipper] Starting service LoaderSrv
281 [D][Flipper] Starting service NotificationSrv
284 [D][Flipper] Starting service PowerSrv
288 [D][Flipper] Starting service StorageSrv
292 [I][Flipper] Startup complete
295 [I][Loader] Executing system start hooks
299 [I][StorageInt] Starting
301 [I][StorageInt] Config: start 080B2000, read 8, write 8, page size: 4096, page count: 35, cycles: 10000
304 [I][StorageInt] Mounted
306 [I][StorageExt] card detected
698 [I][StorageExt] card mounted
701 [I][NotificationSrv] loading settings from "/int/.notification.settings"
712 [E][SavedStruct] Failed to read "/int/.dolphin.state". Error: file/dir not exist
716 [W][SubGhzOnStart] Region data is missing or empty
723 [W][DolphinState] Reset dolphin-state
726 [I][NotificationSrv] load success
759 [E][SavedStruct] Failed to read "/int/.bt.keys". Error: file/dir not exist
765 [E][BtKeyStorage] Failed to read payload size
769 [W][BtSrv] Failed to load bonding keys
773 [I][Core2] C2 boot completed, mode: Stack
777 [I][Core2] Core2: FUS: 1.2.0, mem 16/0, flash 6 pages
780 [I][Core2] Core2: Stack: 1.15.0, branch 0, reltype 3, stacktype 3, flash 31 pages
785 [I][Core2] Radio stack started
788 [I][Core2] Flash activity control switched to SEM7
798 [D][GattChar] Updating Manufacturer Name char
801 [D][GattChar] Updating Serial Number char
804 [D][GattChar] Updating Firmware Revision char
807 [D][GattChar] Updating Software Revision char
810 [D][GattChar] Updating RPC Version char
816 [D][GattChar] Updating Power State char
821 [D][GattChar] Updating RPC status char
825 [D][GattChar] Updating Battery Level char
1071 [I][AnimationManager] Select 'L1_Tv_128x47' animation
1085 [I][SavedStruct] Loading "/int/.desktop.settings"
1477 [I][AnimationManager] Select 'L1_Furippa1_128x64' animation
2482 [I][storage] SD card mount
2485 [I][Loader] Starting StorageMoveToSd
2491 [I][AnimationManager] Unload animation 'L1_Furippa1_128x64'
2507 [I][MoveToSd] Nothing to move
2533 [I][Loader] App returned: 0
2535 [I][Loader] Application stopped. Free heap: 148488
2691 [I][AnimationManager] Select 'L1_Furippa1_128x64' animation
2697 [I][AnimationManager] Load animation 'L1_Furippa1_128x64'
2722 [I][SavedStruct] Loading "/int/.desktop.settings"
5596 [I][Loader] Loading /ext/apps/Sub-GHz/subghz.fap
5719 [I][elf] Total size of loaded sections: 43317
5722 [I][Loader] Loaded in 126ms
5726 [I][AnimationManager] Unload animation 'L1_Furippa1_128x64'
5735 [I][SubGhzSetting] File is not used /ext/subghz/assets/setting_user
5740 [I][SubGhzKeystore] Loading keystore /ext/subghz/assets/keeloq_mfcodes
5795 [I][SubGhzKeystore] Loading keystore /ext/subghz/assets/keeloq_mfcodes_user
5801 [E][SubGhzKeystore] Unable to open file for read: /ext/subghz/assets/keeloq_mfcodes_user
5804 [I][SubGhzProtocoNiceFlorS] Loading rainbow table from /ext/subghz/assets/nice_flor_s
5806 [I][SubGhzProtocoCameAtomo] Loading rainbow table from /ext/subghz/assets/came_atomo
5809 [I][SubGhzProtocoAlutech_at_4n] Loading rainbow table from /ext/subghz/assets/alutech_at_4n
5817 [D][libmgr] Loading /any/apps_data/subghz/plugins/radio_device_cc1101_ext.fal
5883 [I][elf] Total size of loaded sections: 4340
5885 [D][Fap] Library for subghz_radio_device, API v. 1 loaded
5887 [D][Fap] Library for subghz_radio_device, API v. 1 loaded
5890 [I][SubGhzDeviceRegistry] Loaded 2 radio device
6173 [E][SubGhz_Device_CC1101_Ext] Init failed
17912 [D][BrowserWorker] Start
17923 [D][BrowserWorker] Enter folder: /any/subghz items: 4 idx: -1
17925 [D][BrowserWorker] Load offset: 0 cnt: 50
23940 [D][BrowserWorker] End
27695 [D][FuriHalSubGhz] Loading custom preset
27701 [D][FuriHalSubGhz] Reg[0]: 02=0D
27703 [D][FuriHalSubGhz] Reg[2]: 03=07
27705 [D][FuriHalSubGhz] Reg[4]: 08=32
27707 [D][FuriHalSubGhz] Reg[6]: 0B=06
27709 [D][FuriHalSubGhz] Reg[8]: 14=00
27711 [D][FuriHalSubGhz] Reg[10]: 13=00
27713 [D][FuriHalSubGhz] Reg[12]: 12=30
27715 [D][FuriHalSubGhz] Reg[14]: 11=32
27717 [D][FuriHalSubGhz] Reg[16]: 10=17
27719 [D][FuriHalSubGhz] Reg[18]: 18=18
27721 [D][FuriHalSubGhz] Reg[20]: 19=18
27723 [D][FuriHalSubGhz] Reg[22]: 1D=91
27725 [D][FuriHalSubGhz] Reg[24]: 1C=00
27727 [D][FuriHalSubGhz] Reg[26]: 1B=07
27729 [D][FuriHalSubGhz] Reg[28]: 20=FB
27731 [D][FuriHalSubGhz] Reg[30]: 22=11
27733 [D][FuriHalSubGhz] Reg[32]: 21=B6
27735 [D][FuriHalSubGhz] PA[34]:  00
27737 [D][FuriHalSubGhz] PA[35]:  00
27739 [D][FuriHalSubGhz] PA[36]:  00
27740 [D][FuriHalSubGhz] PA[37]:  C0
27742 [D][FuriHalSubGhz] PA[38]:  00
27744 [D][FuriHalSubGhz] PA[39]:  00
27745 [D][FuriHalSubGhz] PA[40]:  00
27748 [D][FuriHalSubGhz] PA[41]:  00
27750 [D][FuriHalSubGhz] PA[42]:  00
27751 [D][FuriHalSubGhz] PA[43]:  00
27760 [D][DolphinState] icounter 2, butthurt 0

[CRASH][Sub-GHz] furi_check failed
	r0 : 200194a8
	r1 : 20010100
	r2 : 1a04201
	r3 : 58000000
	r4 : 1
	r5 : 18
	r6 : 200194a8
	r7 : 20019498
	r8 : a5a5a5a5
	r9 : a5a5a5a5
	r10 : a5a5a5a5
	r11 : 20031330
	lr : 80029d9
	stack watermark: 1584
	     heap total: 182696
	      heap free: 47672
	 heap watermark: 44584
	core2: not faulted
System halted. Connect debugger for more info

[CRASH][ISR 3] HardFault
	r0 : 200194a8
	r1 : 20010100
	r2 : 1a04201
	r3 : 58000000
	r4 : 1
	r5 : 18
	r6 : 200194a8
	r7 : 2002ffc8
	r8 : a5a5a5a5
	r9 : a5a5a5a5
	r10 : a5a5a5a5
	r11 : 20031330
	lr : 8006f83
	     heap total: 182696
	      heap free: 47672
	 heap watermark: 44584
	core2: not faulted
System halted. Connect debugger for more info
```

Skorpionm, 2023-08-24:
please give me the file you are imitating

GLinnik21, 2023-08-24:
Previously, I flashed the device using `DEBUG=1` and ran the command `./fbt flash_blackmagic`. However, now, for some reason, every time I try to access the Sub-GHz app, I'm prompted for a 'Firmware update'. How can I use Sub-GHz after flashing with Blackmagic, especially considering it deletes all the internal configuration files? Updating the firmware seems illogical to me since I'm manually uploading it myself.

skotopes, 2023-08-25:
@GLinnik21 can you provide us file that crashes your flipper?
Also starting from 0.90 an autodetect your programmer, so just use `./fbt flash`

GLinnik21, 2023-08-25:
https://disk.yandex.ru/d/B48OWGyi_sP4uQ

GLinnik21, 2023-08-25:
[Key.sub](https://github.com/flipperdevices/flipperzero-firmware/files/12443821/Key.sub.txt)

Skorpionm, 2023-08-27:
Your file plays correctly. Not throwing any errors. What is your firmware? Does the error reproduce if only SubGhz is launched after HardReset, or does an error occur if another application was launched BEFORE that

skotopes, 2024-01-10:
Please retest with latest dev and reopen if issue persist.

closed 2024-01-10 by skotopes