flipperdevices/flipperzero-firmware #3013: Firmware tar archiving failure in setting uid file property (again)

GLinnik21, 2023-08-27:
### Describe the bug.

While attempting to build the firmware using the `./fbt FORCE=1 flash_usb` command, the build process fails during the tar archiving step. The error indicates a `ValueError` for an "overflow in number field", which seems to be related to setting the `uid` file property in the `tarfile.py` module.

### Reproduction

Execute the command: `./fbt FORCE=1 flash_usb`

### Logs

```Text
Executing task: ./fbt FORCE=1 flash_usb 
APIPREP build/f7-firmware-D/sdk_origin.i
APIPP   build/f7-firmware-D/sdk_origin.i
SDKCHK  firmware/targets/f7/api_symbols.csv
API version 26.3 is up to date
APITBL  build/f7-firmware-D/assets/compiled/firmware_api_table.h
SDKTREE build/f7-firmware-D/sdk_headers/sdk.opts
Linking build/f7-firmware-D as latest built dir (./build/latest/)
CDB     build/f7-firmware-D/compile_commands.json
2023-08-27 23:10:35,781 [INFO] Firmware binaries can be found at:
        dist/f7-D
2023-08-27 23:10:35,783 [INFO] Cannot validate layout for partial package
2023-08-27 23:10:35,784 [INFO] Use this directory to self-update your Flipper:
        dist/f7-D/f7-update-local
Traceback (most recent call last):
  File "scripts/sconsdist.py", line 279, in <module>
    Main()()
  File "scripts/flipper/app.py", line 42, in __call__
    return_code = self.call()
  File "scripts/flipper/app.py", line 58, in call
    return self.args.func()
  File "scripts/sconsdist.py", line 149, in copy
    if bundle_result := self.bundle_update_package():
  File "scripts/sconsdist.py", line 274, in bundle_update_package
    tar.add(bundle_dir, arcname=bundle_dir_name)
  File "python/lib/python3.11/tarfile.py", line 2005, in add
    self.addfile(tarinfo)
  File "python/lib/python3.11/tarfile.py", line 2024, in addfile
    buf = tarinfo.tobuf(self.format, self.encoding, self.errors)
  File "python/lib/python3.11/tarfile.py", line 825, in tobuf
    return self.create_ustar_header(info, encoding, errors)
  File "python/lib/python3.11/tarfile.py", line 844, in create_ustar_header
    return self._create_header(info, USTAR_FORMAT, encoding, errors)
  File "python/lib/python3.11/tarfile.py", line 956, in _create_header
    itn(info.get("uid", 0), 8, format),
  File "python/lib/python3.11/tarfile.py", line 218, in itn
    raise ValueError("overflow in number field")
ValueError: overflow in number field
scons: *** [dist_updater_minpackage] Error 1
```

### Anything else?

macOS version 13.4.1 (c) (22F770820d) 
Python 3.11.4

GLinnik21, 2023-08-27:
Getting this on `dev` branch on commit `9dedcd07b604762a5e8d880567fe9147b2fb2f60`.
The same issue was in #2678

hedger, 2023-08-28:
No wonder you're getting this on ancient commits - 9dedcd07b604762a5e8d880567fe9147b2fb2f60 had been authored before #2678 was reported and fixed.

closed 2023-08-28 by hedger