flipperdevices/flipperzero-firmware #3023: Incorrect regional restrictions for Malaysia

yunaranyancat, 2023-08-30:
### Describe the enhancement you're suggesting.

Currently, Malaysia is listed under 

[image]

However, based on the restrictions by MCMC's class assignment , [pdf](https://www.mcmc.gov.my/skmmgovmy/media/General/registers/cma/Class-Assignment-No-2-of-2022.pdf); it does not allow ;

- 315Mhz (which is included in the list)

[image]

### Anything else?

Nothing

skotopes, 2023-09-04:
@yunaranyancat There are multiple class assignments that can be used for flipper:

- Short-range Radiocommunication Device
- Remote Controlled Device
- Security Device
- Radio Frequency Identification Device
- Industrial, Scientific and Medical Device

Combination of them actually gives even richer band set. We'll talks with our certification lab to clarify which one we applied for and then I'll update configuration.

skotopes, 2024-05-21:
Region was updated according to https://www.mcmc.gov.my/skmmgovmy/media/General/registers/cma/Class-Assignment-No-2-of-2022.pdf, paragraph 11. Security Device

closed 2024-05-21 by skotopes