flipperdevices/flipperzero-firmware #3041: FM1208-10 read does not read more than UID

team-orangeBlue, 2023-09-04:
### Describe the bug.

Using FM1208-10 (Fudan MFC 1K + 7K Smartcard) with flipper does not work with the Mifare Classic part. Instead, the flipper stops immediately and says that the tag is of unknown ISO.

[image]

While you can force a Mifare classic read from extra actions, it appears to be useless and does not find any keys. The tag is keyed using default FF-FF keys, except for key 0A (but it is in the user dictionary): 
[image]

### Reproduction

1. Get a FM1208-10 (they cost 2.8 CNY on taobao and are called "CPU card")
2. Use with proxmark, phone, so on, anything with Mifare Classic support - they read the memory.
3. Use with flipper - instantly quits even though SAK claims support for Mifare classic.

### Target

f-7

### Logs

```Text
433406 [I][DolphinState] State saved
434438 [I][FuriHalNfc] Deinit OK
434566 [I][FuriHalNfc] Init OK
434571 [D][DolphinState] icounter 3731, butthurt 0
434578 [T][FuriHalNfc] State change 0 -> 10
434707 [T][FuriHalNfc] State change 10 -> 20
434879 [T][FuriHalNfc] Timeout
434985 [T][FuriHalNfc] State change 0 -> 10
435112 [T][FuriHalNfc] State change 10 -> 20
435286 [T][FuriHalNfc] Timeout
435390 [T][FuriHalNfc] State change 0 -> 10
435517 [T][FuriHalNfc] State change 10 -> 20
435691 [T][FuriHalNfc] Timeout
435795 [T][FuriHalNfc] State change 0 -> 10
435922 [T][FuriHalNfc] State change 10 -> 20
436096 [T][FuriHalNfc] Timeout
436200 [T][FuriHalNfc] State change 0 -> 10
436335 [T][FuriHalNfc] State change 10 -> 11
436361 [T][FuriHalNfc] State change 11 -> 13
436374 [T][FuriHalNfc] State change 13 -> 30
436419 [I][FuriThread] NfcWorker allocation balance: 0

[ . . . ]
>: nfc detect
Detecting nfc...
Press Ctrl+C to abort
Found: NFC-A UID length: 4, UID:AE0F500B

>: _
```

### Anything else?

Did this happen from the EMV removal change? There probably should be an ATS checker (for FM1208-10 it is `10 78 80 90 02 20 90 00 00 00 00 00 [Tag UID here]`)

doomwastaken, 2023-11-03:
@team-orangeBlue could you check your issue again with latest firmware, we need details after NFC refactor

team-orangeBlue, 2023-11-03:
> @team-orangeBlue could you check your issue again with latest firmware, we need details after NFC refactor

[ Reply ETA approx. 6h30min ]

team-orangeBlue, 2023-11-03:
> @team-orangeBlue could you check your issue again with latest firmware, we need details after NFC refactor

I like this approach...
thank you!
[image]

Hopefully there will be some sort of parsing ATS.

closed 2023-11-03 by team-orangeBlue

gornekich, 2023-11-03:
@team-orangeBlue you can read your tag as ISO14443-4A and there will be parses ATS information in Info screen

team-orangeBlue, 2023-11-03:
> @team-orangeBlue you can read your tag as ISO14443-4A and there will be parses ATS information on Info screen

By that I meant the "Unknown" part.
I tapped some Mifare plus tags, but they didn't have definitions either.