flipperdevices/flipperzero-firmware #3046: Can't read Indala

docpillman, 2023-09-06:
### Describe the bug.

Cannot read Indala. Read RAW and emulate after reading not work too.  Read with amount different firmware version . 

[RAW.zip](https://github.com/flipperdevices/flipperzero-firmware/files/12543049/RAW.zip)
[image]
[image]
[image]
[image]

### Reproduction

1. Read - Not work 
2. Read Indala PSK from extra command - Not work
3. Read RAW work ... But emulation not work 
4. Trying many times ... Reader work ! Card work !

DrZlo13, 2023-09-07:
It's look like the long (128-bit iirc) Indala and it's not currently supported by the Flipper.

docpillman, 2023-09-07:
Why RAW reading and emulation not work ?

docpillman, 2023-09-07:
> It's look like the long (128-bit iirc) Indala and it's not currently supported by the Flipper.

Why RAW reading and emulation not work ?

skotopes, 2023-09-08:
> > It's look like the long (128-bit iirc) Indala and it's not currently supported by the Flipper.
> 
> Why RAW reading and emulation not work ?

Why should it?

RFID RAW was designed for developers to check if payload contains any meaningful data and that is it.
RFID RAW data should never be used for emulation because it is not RAW, but decoded data that lost phase/clock component.

In general we know about lack of Indala 128 support. Sadly we don't have such systems available for testing, so maybe someone from community will implement it.

closed 2023-09-08 by skotopes