flipperdevices/flipperzero-firmware #3047: Can't Patch Firmware

Polar0fficial, 2023-09-07:
### Describe the bug.

In regards to this blog post, "https://techryptic.github.io/2023/09/01/Annoying-Apple-Fans/" it tells me to use this command to patch the firmware "./fbt COMPACT=1 DEBUG=0 VERBOSE=1 updater_package" The part I'm confused about is where to execute that command.

### Reproduction

No idea where to execute this command. If anyone can help, it would be greatly appreciated.

AdminEvilDRagon, 2023-09-07:
I have the same problem/issue

AdminEvilDRagon, 2023-09-07:
I found the problem you need to remove the ./ 
like this

fbt COMPACT=1 DEBUG=0 VERBOSE=1 updater_package

closed 2023-09-07 by hedger

Polar0fficial, 2023-09-09:
I tried that, but I got the following error. "fbt : The term 'fbt' is not recognized as the name of a cmdlet, function, script file, or operable program. Check the
spelling of the name, or if a path was included, verify that the path is correct and try again.
At line:1 char:1
+ fbt COMPACT=1 DEBUG=0 VERBOSE=1 updater_package
+ ~~~
    + CategoryInfo          : ObjectNotFound: (fbt:String) [], CommandNotFoundException
    + FullyQualifiedErrorId : CommandNotFoundException"

I'm supposed to type this in the command prompt application on my PC, right?