flipperdevices/flipperzero-firmware #3051: RFID EM4X50 protocol support

tcfsoft, 2023-09-07:
### Description of the feature you're suggesting.

Can protocol EM4X50 be implemented? 
Adding all info I have, also added proxmark dump in case it helps, in proxmark is recognized as EM4X50.

Flipper raw files:
[dump_ flipper.zip](https://github.com/flipperdevices/flipperzero-firmware/files/12550847/dump_.flipper.zip)

Proxmark dump:
[dump_proxmark.zip](https://github.com/flipperdevices/flipperzero-firmware/files/12550857/dump_proxmark.zip)

### Anything else?

Let me know if I can help with anything else, thank you.

skotopes, 2023-09-08:
@doomwastaken @DrZlo13 do we have em4x tags available for testing?

DrZlo13, 2023-09-08:
The EM4X50 is a recordable blank, not a protocol, and as far as I can tell from the RAW recording, current tag contains empty data. I don't think the Flipper will be able to distinguish this tag from other writable tags.