flipperdevices/flipperzero-firmware #3053: NFC in-app Functionalities

NikoCosmico01, 2023-09-07:
### Describe the enhancement you're suggesting.

When editing HEX dump in-app you cannot see the last bit (out of screen on the right corner).
Implement way to copy/paste whole block.
Suggest to implement "import dump from file" with following formats .mct/.json/.bin/.mdf/.dump/.eml .
Implement way to edit in-app (add or view) user keys (mf_classic_dict_user.nfc).

### Anything else?

Trying to work on it.

skotopes, 2023-09-08:
@NikoCosmico01 can you provide some screenshots, used version and detailed steps to reproduce issue?

NikoCosmico01, 2023-09-08:
> @NikoCosmico01 can you provide some screenshots, used version and detailed steps to reproduce issue?

[image]

Version 1.6.3.1150
No steps needed.