flipperdevices/flipperzero-firmware #3054: Toolchain version is not supported.

ajsdomingues, 2023-09-07:
### Describe the bug.

receive this error ether on macos and on windows 

after this 

git clone --recursive https://github.com/flipperdevices/flipperzero-firmware.git

when I try to compile with ./fbt

get this all the time

scons: *** Toolchain version is not supported. Allowed: (' 10.3.',), toolchain:    Stop.

### Reproduction

git clone --recursive https://github.com/flipperdevices/flipperzero-firmware.git

./fbt

unixispower, 2023-09-08:
I ran into this same issue on a fork of this firmware. I ran `source scripts/toolchain/fbtenv.sh` and then `env` and noticed a parent directory had a space in the filename. Removing the space from the name of the parent directory  allowed me to run `./fbt` without error.

Lixczyy, 2023-09-08:
@unixispower This fixed the issue for me. huge thanks

PR #3105 (2023-09-25): fbt: reworked tool path handling
# What's new

- fbt: removed absolute paths from env setup; moved abs paths to cdb tool
- fixes #3054 
- fbt: pvs: temporarily disabled early license expiration checks

# Verification 

- Clone to path with spaces 
- Check that building firmware works
- Check that vscode config works with clangd

# Checklist (For Reviewer)

- [x] PR has description of feature/bug or link to Confluence/Jira task
- [x] Description contains actions to verify feature/bugfix
- [x] I've built this code, uploaded it to the device and verified feature/bugfix
closed 2023-09-25 by skotopes