flipperdevices/flipperzero-firmware #3072: Sub-Ghz Chat is broken since 0.90.1

twisted-pear, 2023-09-12:
### Describe the bug.

After updating to 0.90.1 reception in the Sub-Ghz chat on the CLI is broken. Messages that are sent are received just fine by a Flipper on something below 0.90.1 but received messages are garbled. Just random characters are printed.

This also affects the ESubGhz Chat app which works effectively the same as the CLI Sub-Ghz chat.

### Reproduction

1. Go to CLI on one Flipper
2. Enter "subghz chat"
3. Go to CLI on second Flipper
4. Enter "subghz chat"
5. Send messages on one Flipper
6. Watch the other flipper receive only a garbled mess.

Skorpionm, 2023-09-12:
SubGhz: Fix CLI subghz chat #3073

twisted-pear, 2023-09-12:
> check #3073

The attached build fixes the chat reception problem. Thank you.

However, it seems to break menu icons in the `gui/modules/menu.h` module. The flipper locks up when a menu item has an icon set that is not an animation but a regular icon. That worked fine before.

It also breaks my background loader app. Again the flipper locks up as soon as the loaded app sends the background loader the message that it's going into background. This is just a furi_message_queue_put().

I'm just letting you know, since this is a dev build. I will open actual bug reports if it happens on a release.

twisted-pear, 2023-09-19:
everything is working in 0.91.4, thank you, closing

closed 2023-09-19 by twisted-pear