flipperdevices/flipperzero-firmware #3074: Impossible to turn off projectors using universal remotes

sashaMSU, 2023-09-12:
### Describe the bug.

The problem is that a lot of projectors require to press Power button twice in a short period of time (5-10 seconds) to be turned off. It is impossible to do so using universal remote with normal projectors.ir list because there each signal is listed only once. Restarting bruteforce will just cause projector timeout if the correct signal is not at the very beginning of the list. I tried duplicating every Power signal in projectors.ir and it solved the problem. I would be glad if you included list with doubled Power signals into firmware.

### Reproduction

1. Point flipper at a working projector
2. Go to universal remotes->projectors
3. Press Power button
4. At about 60% progress projector shows a message asking to press Power again
5. If I let flipper to continue, there is no more fitting signals
6. If I restart the process, it takes too long to reach the correct signal
7. After 10 seconds the message disappears and the projector stays on

### Anything else?

[projectors.zip](https://github.com/flipperdevices/flipperzero-firmware/files/12586181/projectors.zip)
Here is my list with doubled Power signals and a program that I wrote to make it.

skotopes, 2023-09-20:
I think it make sense to double only signals of the projects that needs it. Which one was your?

sashaMSU, 2023-09-22:
I noticed it with InFocus and Epson. Also someone noticed with Optoma HD29HST. But I think this feature is standart for all projectors, because it protects from frequent turning on/off, which is said to be bad for the lamp and to ecology.

skotopes, 2023-09-26:
I still think that we need to do that only if projector requires it. Modern projectors uses LED or laser and don't have such feature.

Leptopt1los, 2023-10-27:
@skotopes maybe this problem can be solved with the following features for a universal remote control: when you press the send button during the bruteforce of signals, the bruteforce should not start from the beginning, but, for example, roll back N signals (let's say N = 10). this functionality will not only solve this problem, but will also allow you to effectively control the volume and channels in tv.ir

sashaMSU, 2023-10-27:
Why not just copy the interface from Subghz bruteforce app? It would make possible to manually iterate through signals, re-send and save them.

skotopes, 2023-10-29:
Once again, why not to duplicate signal that are required to send twice?

Leptopt1los, 2023-10-29:
1. it is difficult to test on the entire set of available signals
2. inefficient file size increase
3. this does not solve the problem of inconvenient volume boost for tv.ir . there is a situation in which we have volume buttons, but it is terribly inconvenient to use them in practice, since we have to wait for the search to the desired signal every time. as a result, the task of increasing the volume by 10 units is already becoming torture. both solutions (mine and @sashaMSU) solve this problem

skotopes, 2023-10-30:
Universal remotes was never intended for everyday use. Just grab from community repo or make copy of your model remote.

I'm closing this ticket:
- We'll think about seek operations in universal remote
- Dumb doubling power button is not going to be accepted
- But we are going to accept doubling for some specific models that requires it

closed 2023-10-30 by skotopes