flipperdevices/flipperzero-firmware #3085: Keeloq unknown: Fantom/Highland T3401 gate remote

wosk, 2023-09-15:
### Description of the feature you're suggesting.

This remote decoded by Flipper as KeeLoq with unknown manufacturer code.
Please add support for it.

[image]

433.92MHz
KeeLoq
FCC ID unknown

Initial link: https://forum.flipper.net/t/fantom-highland-t3401-gate-remote/16804
Also description about ancient S2x4.sub
[image]
Extra video https://youtu.be/EA10QZKUlQ0

[fantom.zip](https://github.com/flipperdevices/flipperzero-firmware/files/12617593/fantom.zip)

skotopes, 2023-09-20:
We are waiting for this remote to arrive to our office.

Skorpionm, 2023-11-08:
I got my hands on this remote control. and I managed to get the encryption key from it, but as always there is a “BUT”. this system uses Normal Learning encryption, that is, to obtain the encryption key, the serial number of the key fob + the manufactory key of the system is used and the resulting key is stitched into the key fob, that’s what I received, that is, it can only be used with a key fob with that serial number and not with the entire system
[image]
that is, the manufactory key is not physically stored in the key fobs of this system, and to obtain it you need to pick the receiver.

wosk, 2023-11-10:
Thanks for your efforts!
According to video https://youtu.be/EA10QZKUlQ0 algorithm is reversible in some kind without access to receiver.

Also in the video receiver R5403 was used.

Skorpionm, 2023-11-11:
They got the man code for the system somewhere.  but their remote control with "Hold the button" works on the principle of re-sync, record through raw 10 consecutive presses of the button with holding for a second and try replaying

tkerby, 2023-12-18:
>  ![image](https://private-user-images.githubusercontent.com/85568270/281320725-b7dbf7a4-dab2-4986-b336-41afc36d1ce3.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTEiLCJleHAiOjE3MDI5MzUxMDYsIm5iZiI6MTcwMjkzNDgwNiwicGF0aCI6Ii84NTU2ODI3MC8yODEzMjA3MjUtYjdkYmY3YTQtZGFiMi00OTg2LWIzMzYtNDFhZmMzNmQxY2UzLnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFJV05KWUFYNENTVkVINTNBJTJGMjAyMzEyMTglMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjMxMjE4VDIxMjY0NlomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPTY0OWMxNTc2ZmQ5YzEyZTJjMTkyZjJmZjA1MWMyYWFiMWZlMWU2ZGI5ZmY5YmE2Y2VhMzliOWQzZGZjYTk5MjkmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0JmFjdG9yX2lkPTAma2V5X2lkPTAmcmVwb19pZD0wIn0._VTyJQAFxjW29Q9cgYH_hWbdjJV2LFg1fTBUFBEDNlo) 

Whats the app you're using there and is it available? I have a few systems I own I'm keen to find manufacturer keys for

Skorpionm, 2023-12-21:
self-written, it is not publicly available

Skorpionm, 2024-04-16:
is not possible, the manufacturing code is different for each key fob

closed 2024-04-16 by Skorpionm