flipperdevices/flipperzero-firmware #3089: [BadUSB Feature Request] Is there a way use the iPadOS "globe" key?

bessx, 2023-09-16:
### Describe the enhancement you're suggesting.

Looking to run a badusb script on iPadOS or iOS, and I was trying to emulate the "globe" (🌐) key. Since there's a usb/bluetooth keyboard for the iPad with a global key, it should be possible to emulate, right?

Is there some shortcut for this, or is this not implemented? Please let me know best practices for this.

Thanks.

Anything else?

skotopes, 2023-09-20:
Which key code this symbol emits?

u1735067, 2023-10-31:
Does this help (sorry for the format, it's kind of personal notes too)?

* https://github.com/KinesisCorporation/Adv360-Pro-ZMK/issues/240
  * https://zmk.dev/docs/codes#application-controls *
  * https://github.com/zmkfirmware/zmk/issues/947
    * **mentions `usage 0x029D`** (this is `AC Next Keyboard Layout Select`)
    * but also "Actually, in Apple keyboards, `fn` and `globe` are physically the same."
    * https://github.com/zmkfirmware/zmk/pull/1938/files
      * https://usb.org/sites/default/files/hut1_2.pdf#page=127 *
        * https://www.usb.org/sites/default/files/hutrr56b_application_keyboard_layout_selector_0.pdf
    * https://developer.apple.com/accessories/Accessory-Design-Guidelines.pdf#page=86 *
* https://gist.github.com/fauxpark/010dcf5d6377c3a71ac98ce37414c6c4
  * (mentions `0x5300`, but it might be an internal value of the qmk software/firmware)
  * https://github.com/qmk/qmk_firmware/issues/2179
    * https://github.com/shiroholloway/HID-Explorer
  * mentions `page 0xFF, usage 0x03` for the FN key, but it seems to be required to announce an Apple keyboard VID/PID?
    * https://opensource.apple.com/source/IOHIDFamily/IOHIDFamily-1035.70.7/IOHIDFamily/AppleHIDUsageTables.h.auto.html
      * `AppleVendor Page Top Case (0x00ff), kHIDUsage_AV_TopCase_KeyboardFn (0x0003)`
  * https://docs.qmk.fm/#/keycodes *
* https://docs.hak5.org/hak5-usb-rubber-ducky/duckyscript-tm-quick-reference
  * https://github.com/hak5/usbrubberducky-payloads/blob/master/languages/us.json (this seems limited?)
* https://github.com/flipperdevices/flipperzero-firmware/blob/0.94.1-rc/applications/main/bad_usb/helpers/ducky_script_keycodes.c
  * https://github.com/flipperdevices/flipperzero-firmware/blob/0.94.1-rc/firmware/targets/furi_hal_include/furi_hal_usb_hid.h
  * https://github.com/flipperdevices/libusb_stm32/blob/6ca2857519f996244f7b324dd227fdf0a075fffb/inc/hid_usage_consumer.h#L409 <- it seems `0x029D` is missing (that would be the `Globe` key), also in `ducky_script_keycodes.c` many of theses are not exposed? Maybe expose way more keys than RubberDucky?
 
\* might contain useful tables?

skotopes, 2023-10-31:
We'll schedule it for next release

skotopes, 2024-01-10:
@nminaylov any progress?

skotopes, 2024-02-18:
@bessx can you test https://github.com/flipperdevices/flipperzero-firmware/pull/3403 ?

bessx, 2024-02-20:
> @bessx can you test #3403 ?

Sure, I'll test it out tonight! Thank you!

closed 2024-07-07 by skotopes