flipperdevices/flipperzero-firmware #3090: Sub-1GHz transmission frequencies in Thailand

sumpo, 2023-09-16:
### Describe the enhancement you're suggesting.

After some extensive research, I conclude that 300-348 MHz and 387-464 MHz are allowed in Thailand [NBTC latest annoucement on allowed frequencies](https://www.nbtc.go.th/getattachment//law/law_noti/nbtc_notification/%E0%B8%9B%E0%B8%A3%E0%B8%B0%E0%B8%81%E0%B8%B2%E0%B8%A8-%E0%B8%81%E0%B8%AA%E0%B8%97%E0%B8%8A-%C2%A0%E0%B9%80%E0%B8%A3%E0%B8%B7%E0%B9%88%E0%B8%AD%E0%B8%87-%E0%B8%AB%E0%B8%A5%E0%B8%B1%E0%B8%81%E0%B9%80%E0%B8%81%E0%B8%93%E0%B8%91%E0%B9%8C%E0%B8%81%E0%B8%B2%E0%B8%A3%E0%B9%83%E0%B8%8A%E0%B9%89%E0%B8%84%E0%B8%A5%E0%B8%B7%E0%B9%88%E0%B8%99%E0%B8%84%E0%B8%A7%E0%B8%B2%E0%B8%A1%E0%B8%96%E0%B8%B5%E0%B9%88%E0%B9%81%E0%B8%A5%E0%B8%B0/%E0%B8%9B%E0%B8%A3%E0%B8%B0%E0%B8%81%E0%B8%B2%E0%B8%A8%E0%B8%AF.pdf?lang=th-TH) with transmitting power no more than 10mW. 920 - 925 MHz are allowed as well with 50mW limitation. I believe that flipper zero should be able to transmit on these frequencies given that it doesn't exceed these rules?

skotopes, 2023-09-18:
@sumpo can you also provide link to official band allocation document?

sumpo, 2023-09-19:
[This](https://www.nbtc.go.th/getattachment/f051ead2-40ae-4c6b-b127-a8eef9b10e2b/%E0%B9%81%E0%B8%99%E0%B8%9A-3.pdf.aspx?lang=th-TH&ext=.pdf) and the one above. Listed under free to operate w/o license and import/export for under 10 mW. There is an older revision of the first doc in English. You may need a translator but the table is straight forward.

skotopes, 2024-05-21:
Region was updated according to https://www.nbtc.go.th/spectrum_management/แผนแม่บท/ตารางกำหนดคลื่นความถี่แห่งชาติ/National-Table-of-Frequency-Allocations-B-E-2555.aspx

closed 2024-05-21 by skotopes