flipperdevices/flipperzero-firmware #3104: Meshtastic app

BrunoKlu, 2023-09-23:
### Description of the feature you're suggesting.

Ability to turn flipper zero into a Meshtascic node using Bluetooth/ meshtastic app / internal or external CC1101

www.meshtastic.org

Stefan2483, 2023-09-30:
or more powerfull external options like EBYTE SX1278? I think there is an external module working on flipper: Rabbit-Labs - Flux Capacitor - Amplified based on this… there are options. So think of sending Mashtastic message from flipper zero or connected phone via bluetooth

skotopes, 2023-10-09:
Sounds like idea for an app and hardware module ;-) 

Also at this point we are not planning such feature.

closed 2023-10-09 by skotopes

grinjerFURRY, 2025-09-21:
😭