flipperdevices/flipperzero-firmware #3127: NFC: Emulate CIPURSE ISO-14443-4, missing ATR

duhow, 2023-10-03:
### Describe the bug.

Related to #1472 , I'm trying to read [T-mobilitat](https://t-mobilitat.atm.cat/) transport card. This is a **CIPURSE** card.
Based on **ATR** / Historical bytes content, reader can extract the "card number" (printed into tag).
Example:

```
3B8B80012A26A7A10C8041 XX XX XX XX ZZ
XX XX = card ID number in hex (eg. `00000605`)
TCK = ZZ (correct checksum) 
```

### Reproduction

- read NFC CIPURSE card `ISO 14443-4 NFC-A`, UID, ATQA 00 44, SAK 20 . no ATR found
- attempt to emulate UID, mobile NFC won't read/detect the card. Tested with NXP TagInfo Android app.
- reading with USB ACR122U appears `ATR: 3B 80 80 01 01`

gornekich, 2023-11-15:
Hello @duhow !
Thanks for the report and sorry for late response.
Recently we released new NFC library and application which I believe fix this issue. Please update your flipper with latest release and try to reproduce this issue

duhow, 2023-12-11:
I see now that Flipper sends the ATR / historical bytes.
It's a good starting point! Thanks!

closed 2023-12-11 by duhow