flipperdevices/flipperzero-firmware #3149: Unsupported gate remote Go Mini

wosk, 2023-10-14:
### Description of the feature you're suggesting.

https://fccid.io/U5Z-GOMN-GOPRMN
868.35 Mhz
Rolling Code
Modulation ASK
IC 1040T46A 1509 OSW

[image]
[image]
[image]
[image]

RAW in AM650
[go_mini.zip](https://github.com/flipperdevices/flipperzero-firmware/files/12904588/go_mini.zip)

Cloning remote https://scimagic.com.cn/products-item/868-35mhz-rolling-code-jcm-go/

skotopes, 2023-10-31:
@Skorpionm what do you think?

Skorpionm, 2023-10-31:
In terms of receiver programming, the system is very similar to BTF.  This is a system with a seed phrase.  record 3 AM650 sends with the remote control transmission frequency.  turn on raw recording, press buttons A+B simultaneously, hold until the LED lights up, release, press and hold A. and so on at least 3 times to enter programming mode

skotopes, 2023-10-31:
@Skorpionm can we this system support to firmware?

Skorpionm, 2023-10-31:
and there must be at least 10 entries for each button and they must be sequential, I can’t guess from 1 button press.  and if there are several remote controls, then records are needed from all

wosk, 2023-10-31:
> and there must be at least 10 entries for each button and they must be sequential, I can’t guess from 1 button press. and if there are several remote controls, then records are needed from all

Files with 10 presses for each button are attached. Also I have a bunch of the another RAW files for it fob.
Unfortunately I don't have another one fob and I don’t have that key anymore.

Skorpionm, 2023-11-08:
[image]

The protocol is precisely dynamic, Manchester encoded data. Yes, we can accept this protocol and decode the package, but it will not make any sense if we cannot decrypt it. And without a keychain in my hands, I don’t know how to do it

Skorpionm, 2023-11-08:
x3F071228E673431A64D9F01596
x3FF8EDD7641ACF45608E2DAF9A
x3F0712289BE530BA9F270DA500
x3F0712285730C694A0725792DF
x3FF8EDD72A39F9121854D9CDA2
x3F071228D5C606EDE7AB2632A2
x3F0712283636683F5F3591113F
x3F071228D5C606EDE7AB2632A2
x3F0712282AEB6CBC1B5F112D94
x3FF8EDD7E32059E265A196D9F1
It’s not clear which buttons you pressed, in what sequence, whether all the parcels were captured (some of them were broken)
but I’ll say for sure the parcels consist of 13 bits of data, Manchester 1000s/sec

Skorpionm, 2024-04-16:
I can’t decipher with only notes

closed 2024-04-16 by Skorpionm

desson-ctrl, 2024-09-17:
@Skorpionm is there something else you need for this?

My condo recently switched to these remotes. Mind you not the pro version. I can get 10 key presses from both buttons on my remote, plus a few of my neighbours as well. Is that helpful?

Skorpionm, 2024-09-18:
No, it won't help. It's very difficult to understand rolling protocols without having a device at hand.