flipperdevices/flipperzero-firmware #3150: Clock and Icons are drawn in PowerOff and Up Arrow Menus.

LeeroysHub, 2023-10-15:
### Describe the bug.

If you have the clock and icons enabled on the Desktop, the power off and up arrow menu both still display the clock over the screen. This does not look good!

I added this code at the start of the gui_redraw_status_bar routine, because the hide_statusbar_count is not being used in the draw routines at all. Clearly this was started and not finished, but this final addition makes it all look great!

Change is in gui.c

static void gui_redraw_status_bar(Gui* gui, bool need_attention) {
    ViewPortArray_it_t it;
    uint8_t left_used = 0;
    uint8_t right_used = 0;
    uint8_t left_slim_used = 0;
    uint8_t right_slim_used = 0;
    uint8_t top_used = 0;
    uint8_t width;

    //We dont want to draw the status bar if its hidden.
    if(gui->hide_statusbar_count > 0) {
        return;
    }

### Reproduction

1. Enable Clock on Desktop
2. Hold Back to poweroff, view clock on screen.
3. Press Back, press Up to show clock on menu there too.

doomwastaken, 2023-10-25:
Could not reproduce, what firmware version are you using?

LeeroysHub, 2023-10-27:
Sorry it was a CFW issue!

LeeroysHub, 2023-10-27:
I will close!

closed 2023-10-29 by hedger