flipperdevices/flipperzero-firmware #3162: NTAG215 Write

jlaughter, 2023-10-20:
### Description of the feature you're suggesting.

Ability to write saved/generated NTAG215 tags.  While emulating into a cellphone and then wroting is a workaround today, workflow would be streamlined by writing directly.

skotopes, 2023-10-31:
Like writing not to the same card but different one?

jlaughter, 2023-10-31:
Basic write, erase, and lock/unlock of writeable tags/stickers/business cards.   Same card, different card, whatever as long as it’s writeable.  Thanks!

kirkpe, 2023-11-02:
How can I access this?

skotopes, 2023-11-02:
@jlaughter same card write should be available out of the box. Writing to magic cards(various mifare formats) available in NFC Magic app. Writing to different card with different UID will be available later.

kirkpe, 2023-11-02:
Can I clone mifare 1k classic cards? Like copy the data from one card and write to another?

skotopes, 2023-11-02:
> Can I clone mifare 1k classic cards? Like copy the data from one card and write to another?

yes. Full clone is only possible with a magic card. Partial clone requires manual actions(copy pasting data you want to destination card dump and writing it back)

jlaughter, 2023-11-02:
I think there's a disconnect here.  This issue is specifically for NTAG215 tags - URLs, contact info, amiibos, etc.  NOT Mifare. 
 
For NTAG215, emulation is possible, but there's no write capability.   Most mobile phones support NTAG write, so there's a workaround, but it's messy.

jlaughter, 2023-11-02:
Just added screenshots to original request, linked here:

[image]
[image]
[image]

gornekich, 2023-11-15:
Hello @jlaughter !
Took that into work. I will let you know when writing will be implemented

gornekich, 2023-12-05:
Initial NTAG write came with #3246 . Now flipper can write NTAG213/215/216. The conditions are that both source and destination NTAGs are not locked. So they both have default PASSWORD and PACK and the lock bits are reset. In other case the logic of writing becomes very complicated.

Please, test this feature in lates RC or dev. Soon it will be on release. I will close this issue for now, please reopen it if you have any problems with writing data on unlocked NTAGs. If you have any suggestions on improving this feature, please open another issue with feature request.

closed 2023-12-05 by gornekich