flipperdevices/flipperzero-firmware #3168: NFC: Certain DESFire cards cannot be read

gsurkov, 2023-10-27:
### Describe the bug.

According to @noproto's data, two MF DESFire EV2 cards cannot be read. Attempting to read them also leads to application abnormal behaviour.

Original posts:
- [First mention](https://github.com/flipperdevices/flipperzero-firmware/pull/3050#issuecomment-1752125993)
- [Logs with additional debug output](https://github.com/flipperdevices/flipperzero-firmware/pull/3050#issuecomment-1771718606)
- [Proxmark logs](https://github.com/flipperdevices/flipperzero-firmware/pull/3050#issuecomment-1774233772)

More info is available in the of #3050 discussion thread.

### Reproduction

The bug could not be reproduced by any of the devs.

sodoku, 2023-10-30:
Not sure if it's the same issue, but I can't read one of my Desfire cards with `0.94.1-rc` it hangs. It works fine with `0.93`.

The log says:

```
311686 [D][ST25TBPoller] error during trx: 2
311719 [D][Nfc] FWT Timeout
311744 [D][Nfc] FWT Timeout
311769 [D][Nfc] FWT Timeout
311821 [D][Nfc] FWT Timeout
311849 [D][Nfc] FWT Timeout
311851 [D][ST25TBPoller] error during trx: 2
311920 [D][Nfc] FWT Timeout
311945 [D][Nfc] FWT Timeout
312008 [D][Nfc] FWT Timeout
312047 [D][Nfc] FWT Timeout
312049 [D][ST25TBPoller] error during trx: 2
312074 [D][NfcScanner] Found 4 children
312131 [D][Nfc] FWT Timeout
312133 [D][Nfc] FWT Timeout
312167 [D][Nfc] FWT Timeout
312200 [D][Iso14443_4aPoller] Read ATS success
312221 [I][NfcScanner] Detected 1 protocols
312353 [D][Iso14443_4aPoller] Read ATS success
312360 [D][MfDesfirePoller] Read version success
312365 [D][MfDesfirePoller] Read free memory success
312368 [D][MfDesfirePoller] Read master key settings success
312372 [D][MfDesfirePoller] Read master key version success
312376 [D][MfDesfirePoller] Read application ids success
312381 [E][MfDesfirePoller] Failed to read applications
312384 [D][Nfc] FWT Timeout
312386 [D][Nfc] FWT Timeout
312489 [D][Nfc] FWT Timeout
```

skotopes, 2023-10-31:
Yep we received those card, they are clearly differ from the standard ones. We'll include fix in next release.

Jupiops, 2023-11-01:
I have a NXP MIFARE DESFire Ev2 that also just in some rare cases get's read successfully by the Flipper Zero as `ISO 14443-4A (Unknown)` but in 90% of the time it just hangs in `Reading card, Don't move...`. The card is a danish public transport card called [Rejsekort](https://www.rejsekort.dk/)

Here is the proxmark read log of the card:
```
[usb] pm3 --> hf mfdes info

[=] ---------------------------------- Tag Information ----------------------------------
[+]               UID: 04 32 25 8A DE 0F 90 
[+]      Batch number: CF 5D 14 41 60 
[+]   Production date: week 44 / 2021
[+]      Product type: MIFARE DESFire native IC (physical card)

[=] --- Hardware Information
[=]    raw: 04010112001805
[=]      Vendor Id: NXP Semiconductors Germany
[=]           Type: 0x01
[=]        Subtype: 0x01
[=]        Version: 12.0 ( DESFire EV2 )
[=]   Storage size: 0x18 ( 4096 bytes )
[=]       Protocol: 0x05 ( ISO 14443-2, 14443-3 )

[=] --- Software Information
[=]    raw: 04010102011805
[=]      Vendor Id: NXP Semiconductors Germany
[=]           Type: 0x01
[=]        Subtype: 0x01
[=]        Version: 2.1
[=]   Storage size: 0x18 ( 4096 bytes )
[=]       Protocol: 0x05 ( ISO 14443-3, 14443-4 )

[=] --------------------------------- Card capabilities ---------------------------------

[=] --- Tag Signature
[=]  IC signature public key name: DESFire Ev2
[=] IC signature public key value: 04B304DC4C615F5326FE9383DDEC9AA8
[=]                              : 92DF3A57FA7FFB3276192BC0EAA252ED
[=]                              : 45A865E3B093A3D0DCE5BE29E92F1392
[=]                              : CE7DE321E3E5C52B3A
[=]     Elliptic curve parameters: NID_secp224r1
[=]              TAG IC Signature: 9BB3F701D053A9D5DCD871A0A0DD987C
[=]                              : F0F5C02AB5CA9012F858BC5F13B80FD3
[=]                              : 7EEC060589F9DBD725DB5F1CA6452CF5
[=]                              : 6CD05B42E5AF5F3D
[+]        Signature verification: successful

[+] --- AID list
[+] AIDs:  7d0005

[+] ------------------------------------ PICC level -------------------------------------
[+] Applications count: 1 free memory 1024 bytes
[+] PICC level auth commands: 
[+]    Auth.............. NO
[+]    Auth ISO.......... NO
[+]    Auth AES.......... YES
[+]    Auth Ev2.......... YES
[+]    Auth ISO Native... YES
[+]    Auth LRP.......... NO
[+] PICC level rights:
[+] [1...] CMK Configuration changeable   : YES
[+] [.0..] CMK required for create/delete : YES
[+] [..1.] Directory list access with CMK : NO
[+] [...1] CMK is changeable              : YES
[+] 
[+] Key: AES
[+] key count: 1
[+] PICC key 0 version: 1 (0x01)

[=] --- Free memory
[+]    Available free memory on card         : 1024 bytes

[=] Standalone DESFire
```

bettse, 2023-11-11:
I have a Desfire EV1 from KISI that also has the same infinite `Reading card, Don't move...`.  Here are the logs:

```
6549766 [D][RpcGui] SendInputEvent
6549824 [D][RpcGui] SendInputEvent
6549833 [D][NfcScanner] Found 5 base protocols
6549841 [D][DolphinState] icounter 11, butthurt 0
6549884 [D][Nfc] FWT Timeout
6549905 [D][RpcGui] SendInputEvent
6549909 [D][ViewDispatcher] View changed while key press 2001CCC8 -> 2001CEB0. Sending key: OK, type: Release, sequence: 00000035 to previous view port
6549912 [D][Nfc] FWT Timeout
6549963 [D][Nfc] FWT Timeout
6549991 [D][Nfc] FWT Timeout
6549993 [D][ST25TBPoller] error during trx: 2
6550017 [D][NfcScanner] Found 4 children
6550074 [D][Nfc] FWT Timeout
6550077 [D][Nfc] FWT Timeout
6550110 [D][Nfc] FWT Timeout
6550144 [D][Iso14443_4aPoller] Read ATS success
6550176 [I][NfcScanner] Detected 1 protocols
6550303 [D][Iso14443_4aPoller] Read ATS success
6550311 [D][MfDesfirePoller] Read version success
6550314 [D][MfDesfirePoller] Read free memory success
6550318 [E][MfDesfirePoller] Failed to read master key settings
6550321 [D][Nfc] FWT Timeout
6550323 [D][Nfc] FWT Timeout
6550425 [D][Nfc] FWT Timeout
6550528 [D][Nfc] FWT Timeout
6550631 [D][Nfc] FWT Timeout
6550734 [D][Nfc] FWT Timeout
6550837 [D][Nfc] FWT Timeout
6550940 [D][Nfc] FWT Timeout
6551043 [D][Nfc] FWT Timeout
6551146 [D][Nfc] FWT Timeout
```

LeeroysHub, 2023-11-26:
Myki DESFire Mini 0.3K used to work before refactor, but now Myki public transport card wont read either. Stays on Reading card, Don't move. Don't have proxmark to dump unfortunately!

skotopes, 2023-12-07:
@LeeroysHub fix is coming soon

jkter, 2023-12-18:
With firmware 96.1 the problem with `Reading card, Don't move...` infinite loop still occurs.

According to my observations with several cards it seems that the infinite loop occurs when there is an application with unavailable key configuration present on the card.

I could read empty card without any problems. After adding an application with publicly unavailable key configuration (in my case 2N PICard), flipper gets stuck in an infinite loop while reading.

skotopes, 2023-12-18:
@jkter probably same issues, please wait till fix will be released

gsp8181, 2023-12-27:
I've got a few other DESFire cards that don't work

Ola money (Mumbai Metro)
Rabbit (Bangkok Stored Value Card)
PTV Myki Card (Melbourne), this loops with or without the parser installed

kautzz, 2023-12-29:
Can confirm NXP DESFire EV3 stuck on „reading card don’t move…“ with fw version 0.97.1

skotopes, 2023-12-30:
@kautzz yep we are working on it. fix coming soon.

@gsp8181 do you have them? can you check them with proxmark?

gsp8181, 2023-12-30:
@skotopes sure thing

bobbylapointe-ops, 2024-03-15:
Same problem here, with Xtreme or Momentum...

gornekich, 2024-04-16:
The fix arrived in dev. Please, reopen the issue if the problem persists.

closed 2024-04-16 by gornekich

pelrun, 2024-09-11:
Just tried to read my PTV Myki card again after installing 1.0.1 and the behaviour is unchanged. Log shows reads constantly failing with the following error:

41589083 [E][MfDesfirePoller] Failed to read free memory

I've doublechecked that the previous fix attempt made it into the release, and it appears it has, but `mf_desfire_poller_read_free_memory()` is returning with some other error code than `MfDesfireErrorNotPresent` so it still bombs out.

rofessori, 2025-04-19:
Same issue now with 1.2, what was wrong the last time?