flipperdevices/flipperzero-firmware #3171: Mifare Classic 1k emulate unstable?

HectorLee1992, 2023-10-29:
### Description of the feature you're suggesting.

Hi everyone, sorry for my poor English, today I took my Flipper to NFC reader in convenience stores, the reader couldn't read the Mifare classic card that device emulated.(I try more than 10 times in different stores).
I know Flipper zero need to emulate not only the UID but also the whole content of Mifare classic card if I want to use it to recharge money by the NFC reader in the stores.
My question is the latest firmware still can't emulate the whole Mifare Classic 1K card or not?
Or the problem I encountered is related to ATQA or SAK problem? (I know there is vanity SAK and real SAK in this [doc](https://gist.github.com/equipter/3022aea4e371e585ff6e46de637e7769))

BTW, I can recharge the money in the train station with recharge mechaine by Flipper zero once. 
However, It is usually rejected when recharging money.

hedger, 2023-10-29:
Flipper's NFC emulation works in current release. So it did in older ones. However, it's not perfect, because some readers do emulation detection tricks or something non-standart. Additionally, they sometimes do validation of data on the backend, so you cannot just clone the card and use both original an emulated one.

If you encounter a reader that does not accept Flipper's emulation when all sectors have been read, but works with an original card - try getting transaction logs from both Flipper's emulation and original card with Proxmark, then share those logs so it would be possible to find the difference in emulation.

closed 2023-10-29 by hedger

HectorLee1992, 2023-10-30:
Thanks @hedger, I know some system monitor clone card with their backend.
Due to some reasons, I don't have the original card but I get the card dump file. 
Do you have any suggestion?