flipperdevices/flipperzero-firmware #3172: SubGHz: Read Raw misses some transmitted Data

AlexHuebi, 2023-10-30:
### Describe the bug.

When I set my Flipper to "Read RAW", it always misses the first bit of data. This effect is amplified tenfold when the RSSI Threshold is set. It is in all Versions. I've tested it until the latest RC 0.94.1 and on the R01 and R04 Variants.

To sanity check myself, I've also borrowed an Ettus Research SDR to check if the Data is really transmitted, which is the case. So the issue lies within the Flipper itself.

Note: All relevant Files are in the following archive: [recordings.zip](https://github.com/flipperdevices/flipperzero-firmware/files/13205887/recordings.zip)

The Recordings contain the transmission of a Weather Station. To Decode the Data with the Lab Website, use PPM, Short 2000, Long 1000, Sync 4000. To make the Sync Pulses visible use 1000 for both Short and Long and 2000 for Pulse. The Protocol can be decoded with the Weather Station App but that won't show the missing Data at the beginning.

The Packets are transmitted 12 times and are 36bit long, but when I let the Lab Website analyze the Data it shows that parts of the first transmission are corrupted: 

```
              __ __
Bits: {573} / FE D0 EB F3 40 / 3E 90 EB F3 40 / 3E 90 EB F3 40 / [...]
```

When a RSSI is set, the first packet is completely unusable: 

```
Bits: {540} 2A 80 / 3E 90 E8 F3 50 / 3E 90 E8 F3 50 / [...]
```

With other devices, this data loss does result in the inability to replay the recorded files. 

Also included is a generated.sub file, which can be used to test this issue when a second Flipper Zero sends this file. The Recordings, that contain '-generated' in the Filename, are from this file. 

### Reproduction

.

closed 2023-10-30 by skotopes

AlexHuebi, 2023-10-30:
*IF* I read the datasheet of the CC1101 correctly, it should support something called gain freeze. That might help with the gain freeze. So if there would be a Learning Transmission, then it should help with that kind of issue.

skotopes, 2023-10-31:
@AlexHuebi Before gain can be frozen it must be adjusted. Once again there is not much that we can do on our side. Yes it is possible to adjust AGC configuration, but it will sacrifice something else.

AlexHuebi, 2023-10-31:
Yeah I know. I was thinking that there may be a second option or somewhat where the flipper doesn't record the signal but looks at the RSSI. If the RSSI is stable (or as stable as it'd get), AGC gets frozen. Then the second transmission shouldn't overshoot again. Would that not be feasible?

skotopes, 2023-10-31:
You want some SDR functionality ;-) It's coming in next flipper generation

AlexHuebi, 2023-10-31:
Huh, now you made me curious. Is there anything else you could tell me about that? 
Kinda want to know if I should put some money to the side to buy it when its released ^^.

skotopes, 2023-10-31:
We'll announce when it will be ready )