flipperdevices/flipperzero-firmware #3174: Bluetooth RPC disconnects with timeout during data transfer

liamhays, 2023-10-30:
### Describe the bug.

I'm working on a desktop application to transfer files to the Flipper over Bluetooth RPC. I can write commands and data to the Flipper just fine, but receiving data is iffy at best. For example, if I send a valid `StorageReadRequest`, correctly encoded as a length-delimited byte string, the Flipper will send a little bit of `StorageReadResponse` data before disconnecting with reason 08 (disconnect due to supervision timeout). The computer sits in a loop, waiting for indications, and I know that it is successfully ACKing the data with `log trace` on the Flipper. I don't know why the connection is timing out, because data is definitely being transferred, which should reset the connection timeout counter.

Flipper log of the entire connection: (I added the `remained:` logs in `serial_service.c`)

```
305859 [I][BtGap] Connection parameters: Connection Interval: 36 (45 ms), Slave Latency: 0, Supervision Timeout: 42
305963 [D][BtGap] Slave security initiated
306143 [I][BtGap] Pairing complete
306150 [I][BtSrv] Open RPC connection
306154 [D][GattChar] Updating Flow control char
306159 [D][RpcSrv] Session started
306162 [D][GattChar] Updating RPC status char
306168 [D][GattChar] Updating Battery Level char
306233 [I][BtGap] Rx MTU size: 414
306954 [D][BtSerialSvc] Received 31 bytes
306956 [D][BtSerialSvc] Available buff size: 993
306959 [D][RpcStorage] Read
306960 [T][StorageApi] File/Dir 0000E6B0 alloc
306966 [T][StorageApi] File 0000E6B0 - 000101EC open (/ext/apps/NFC/picopass.fap)
306972 [I][BtSerialSvc] remained: 168
306974 [I][BtSerialSvc] remained: 0
307670 [I][BtGap] Disconnect from client. Reason: 08
307674 [I][BtSrv] Close RPC connection
```

Timeout is caused by the BLE connection supposedly not transferring any data, which makes no sense, because data is obviously in transit. 

I think it's likely that this is caused by my code either being buggy or not correctly getting data from the Flipper. Is there a correct way to connect to the Flipper over Bluetooth, or documentation for using RPC over Bluetooth? Everything I've done so far has been derived from existing source, and I find the source code of the mobile apps very difficult to read and navigate (because of the code layout that the apps have to use).

### Reproduction

Basically:

1. Connect to the Flipper over BLE
2. Send the encoded StorageReadRequest to the RX characteristic
3. Subscribe to the TX characteristic and wait for notifications
4. You may get one or two notifications, but the Flipper will then disconnect

If it's helpful, I can put the work-in-progress client code in an accessible repository.

skotopes, 2023-10-30:
Please check ios or android source code: there is a flow control that must be properly implemented.

closed 2023-10-30 by skotopes

liamhays, 2023-11-18:
Adding a comment (not reopening) here in case anybody else comes across this: although the flow control characteristic is important, it's also only for transmitting *to* the Flipper. The issue I was having is not caused by a problem with the Flipper, it's caused by my specific Intel Bluetooth card. See the code and README [here](https://github.com/liamhays/flipwire).