flipperdevices/flipperzero-firmware #3182: Add new SubGhz protocol for Clemsa MasterCode MV12

flipperzelebro, 2023-11-01:
### Description of the feature you're suggesting.

I have a Clemsa Mastercode MV12 garage remote that its not supported now in flipper zero. 
Its a fixed code 433.92 MHz remote with modulation AM270.

Testing with flipperzero i have found that its very similar to Clemsa protocol already implemented, but with some changes:

- Time of the On, Off, pause.
- Length of the key.
- Order of the bits for switches.  

I attach the SUB files for te button 1 and button 2.
The DIP configuration is :
  
(+): ---**---
(o): --------
(-): \*\*\*--\*\*\*

I have already implemented the new protocol and could create a MR

[Mv12_B1.zip](https://github.com/flipperdevices/flipperzero-firmware/files/13226175/Mv12_B1.zip)

[image]

[image]

### Anything else?

I have already implemented the new protocol and could create a MR

skotopes, 2023-11-01:
Yes please, we are waiting for your PR.

UrSuLa360, 2023-11-01:
Looking forward to try it with my clemsa remote.

racdamico, 2023-11-06:
Sure. Yes please. I'm interested too.

UrSuLa360, 2023-11-24:
Flipper detects the signal from the remote and save it correctly.

But emulation does not open the door.

I have attached 4 files: Button 1 raw and Button 1 parsed. And same for button 2.

[image]
[stuff.zip](https://github.com/flipperdevices/flipperzero-firmware/files/13461361/stuff.zip)

Skorpionm, 2023-11-25:
@UrSuLa360  check plz https://github.com/flipperdevices/flipperzero-firmware/pull/3244

UrSuLa360, 2023-11-25:
@flipperzelebro first of all thanks for your contribution. I was looking forward for this since long ago.

May i ask you if  you have tested your code with flipper on a garage door successfully?

Maybe im doing something wrong.

flipperzelebro, 2023-11-25:
Hello @UrSuLa360  , you are wellcome.  Yes I tested it on my own garage. 
I was reviewing the SUBs you shared and I realised yours are sligtly diferent form mine. Your Signal durations are about 5% shorter than mine. I am going to modify the program and check if with new durations the Flipper Zero emulates yours. Hope it wont takes me long

flipperzelebro, 2023-11-25:
In fact the PR @Skorpionm did [#3244 ](https://github.com/flipperdevices/flipperzero-firmware/pull/3244)  is almost the one I was  testing.  

I have tested with 
static const SubGhzBlockConst subghz_protocol_mastercode_const = {
    //.te_short = 1072,
    //.te_long = 2145,
    //.te_delta = 150,
    .te_short = 1020,
    .te_long = 2050,
    .te_delta = 200,
    .min_count_bit_for_found = 36,
};
and the signal played is almost identical to yours. 
Could yo try it?

UrSuLa360, 2023-11-25:
No luck, the door doesn't respond.

I also have another remote, it is not from clemsa brand, but it opens the same doors.

I have attached some pictures and another raw recordings.

The recording **newremote.sub** contains:

long press button 1 new remote
log press button 1 new remote
long press button 2 new remote
long press button 2 new remote

**2remotes.sub** contains:

long press button  clemsa
long press button  clemsa
long press button  new remote
long press button  new remote
(i think the signals from cleamsa open door 1 and signals from newremote open door 2)

Both remotes open the same doors.

Hope this helps you.

PD: The newremote is not detected in flipper. Only the clemsa one is detected.

[2ndremote.zip](https://github.com/flipperdevices/flipperzero-firmware/files/13465426/2ndremote.zip)

Skorpionm, 2023-11-26:
some kind of mysticism.  should have worked as in the first option.  since a 5% difference in durations is very small for such systems.  But are you sure that the remote control is not 2-band?  and tried to bring the flipper close to the receiver

UrSuLa360, 2023-11-26:
There are only 2 remotes from clemsa with 2 buttons and this protocol.
Old one(deprecated)
http://www.remote-control-esma.com/clemsa-mastercode-mv-12-remote-control/
Updated version.
http://www.remote-control-esma.com/clemsa-mastercode-mv-12-d-remote-control/

They both work with the same garage doors. 

I attach a photo from mine without the battery.

[image]

I have a question @flipperzelebro , can you open your garage door from a **raw** recording?

Today im going to test emulating the code very close to the receiver. I was trying from 2 meters before.

Skorpionm, 2023-11-26:
Can you take a photo of the insides of the board of this key fob, so that you can see the tracks and markings of the parts, it is unlikely that they are fsk, but still

UrSuLa360, 2023-11-26:
https://github.com/flipperdevices/flipperzero-firmware/assets/125646802/4b42077c-d341-4e61-b470-61e8ae9f7f87

Here it is a video of the test, as you can see only the real remote works. After the video i tried very very close to the receiver with no luck. Also the replay attack(raw recording) did not work (this is extremely weird imo). 

I have just read your last message, i will prepare the photos soon.

flipperzelebro, 2023-11-26:
I have been studying the signals from the SUB files and what I can see is that the durations of the new remote are longer than the original clemsa, aprox. 15% longer.
Clemsa :          short 1022 / long 2044 / stop 15300
New Remote : short 1200 / long 2350 / stop 20300 
 Also what I see is that the new remote is sending 48 bits (B7E08AAC8B7E) versus the 36 bits (B7E08AAC4) of the original. It looks like the new one has saved part of the start of the next frame as the end of the current one. It is surprising that it works with such different signals.

I have double checked with the values:
.te_short = 1022,
.te_long = 2044,
.te_delta = 200,
and the signal generated is 100% identical to yours.  Somethig is mising I am sure.

I Tried reply signal with Raw and it work for me.

UrSuLa360, 2023-11-26:
I think the problem is if my raw recordings don't open the door, any generated signal from them won't work too.
I need to achieve a working raw recording. 
If flipper can't find a way to mimic a signal from a static code sent from that remote im afraid this gonna be hard...

flipperzelebro, 2023-11-27:
Have you tried recording the raw data with RSSI Threshold? I usually set RSSI Threshold to -70 dBm and signal is much clearer

UrSuLa360, 2023-11-30:
I tried with RSSI threshold -70dBm and ALL modulation built in flipper (AM270, AM650, FM238, FM 476)
no success :(
I combined all available options to record the raw signal, even with other frequencies. No luck.

I have bought a RTL-SDR dongle. I read an old comment from @skotopes in other threads that using this device the signal can be recored and analyzed much better with SDLSharp software.
So i just need time to figure out and research how achieve this task.
I will do my best to help to solve this mystery, just give me time.

And again thank you all for your work in this project.

PD: i know @skotopes you asked for some photos from the clemsa remote board. In order to access the board i need to break the case, so i will try to exhaust all possible options before breaking it. Anyway you check the photos from the board of the "2nd remote" in [this comment 
](https://github.com/flipperdevices/flipperzero-firmware/issues/3182#issuecomment-1826439700)

aluvare, 2023-12-18:
I have done tests with my garage, the message decoding works perfectly, but the garage does not open when you use what has been captured, not even in RAW on AM270.

cristofer1989, 2024-01-19:
Hello.
Same problem here. Flipper decodes the message but it did't work.
Its weird because I can copy the remote even with a "universal" one from Aliexpress but not with the Flipper.

njoythegame, 2024-01-27:
Any news regarding this topic? Has anyone managed to use the flipper zero to open Clemsa doors?

UrSuLa360, 2024-01-28:
[RTL-SDR-clemsa-3signals-433_92MHz-1MSps-1MHz.zip](https://github.com/flipperdevices/flipperzero-firmware/files/14077583/RTL-SDR-clemsa-3signals-433_92MHz-1MSps-1MHz.zip)
 
A promise is a promise.
Here there are the samples recorded with rdl-sdl dongle and URH. (3 times pressed the button)

I saved them in WAV and c16complexsigned. Let me know if you need in another format.

Here there is a signal preview.

[image]

@Skorpionm

UrSuLa360, 2024-03-08:
Any update on this issue?
Is there anything i can do to help?

skotopes, 2024-03-12:
@UrSuLa360 other than creating PR with fix, only wait for @Skorpionm to get some free time

Skorpionm, 2024-04-16:
Even so, I assume that these are different protocols under the same name.  If the issue is resolved I will close the issues

skotopes, 2024-07-07:
@UrSuLa360 was this issue resolved?

UrSuLa360, 2024-07-11:
@skotopes the garage door doesn't open with the decoded signal. Other users report the same problem.
Feel free to close the issue as noone is working on it.

Skorpionm, 2024-07-12:
do they work simply with recorded Raw signal? maybe just wrong frequency is used

UrSuLa360, 2024-07-15:
No luck with raw recordings. Tested ALL combinations of frequencies and modulations.
It is a fixed code protocol that flipper can't replay yet.
I attached some recordings in this thread (using flipper and a rdl-sdr dongle), maybe anyone can find any clue  about what is happening.

Skorpionm, 2024-07-17:
if the receiver does not receive a raw signal, and the code is static, then 100% something is wrong either with the modulation or with the frequency. show the signal spectrum via SDRSharp

UrSuLa360, 2024-07-17:
> [RTL-SDR-clemsa-3signals-433_92MHz-1MSps-1MHz.zip](https://github.com/flipperdevices/flipperzero-firmware/files/14077583/RTL-SDR-clemsa-3signals-433_92MHz-1MSps-1MHz.zip)
> 
> A promise is a promise. Here there are the samples recorded with rdl-sdr dongle and URH. (3 times pressed the button)
> 
> I saved them in WAV and c16complexsigned. Let me know if you need in another format.
> 
> Here there is a signal preview.
> 
> ![clemsa_preview](https://private-user-images.githubusercontent.com/125646802/300291311-29cd57d0-f0bf-44e4-9dda-b66e565a6015.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MjEyMDI5MjIsIm5iZiI6MTcyMTIwMjYyMiwicGF0aCI6Ii8xMjU2NDY4MDIvMzAwMjkxMzExLTI5Y2Q1N2QwLWYwYmYtNDRlNC05ZGRhLWI2NmU1NjVhNjAxNS5wbmc_WC1BbXotQWxnb3JpdGhtPUFXUzQtSE1BQy1TSEEyNTYmWC1BbXotQ3JlZGVudGlhbD1BS0lBVkNPRFlMU0E1M1BRSzRaQSUyRjIwMjQwNzE3JTJGdXMtZWFzdC0xJTJGczMlMkZhd3M0X3JlcXVlc3QmWC1BbXotRGF0ZT0yMDI0MDcxN1QwNzUwMjJaJlgtQW16LUV4cGlyZXM9MzAwJlgtQW16LVNpZ25hdHVyZT1kNjE2NDFlNjg3ZDE1MmEyMmNmYTIzZmQyOWUyNjQxZGZlMzc1NTAzMDA2MTg5YjhhZjhhY2IwYmJmNzM5Nzc5JlgtQW16LVNpZ25lZEhlYWRlcnM9aG9zdCZhY3Rvcl9pZD0wJmtleV9pZD0wJnJlcG9faWQ9MCJ9.vSMhJD2xQA_F_0aXLJFBIOfU2BlqKsXGoQOi0bHO2LI)
> 
> @Skorpionm

Here is some raw recorded signals with URH maybe useful, i wil install SDRSharp and give what you asked soon. Thanks

OscarAkaElvis, 2024-08-21:
Hi, any update on this? I have a Clemsa Mastercode MV1 (instead of MV12) but my case is exactly the same. I'm not able to open the garage door using any of the available options. I tried RAW using all combinations and modulations. I can get the mastercode and the map got of the dip switches is reflecting well the physical dip switch positions, but is not working in any case.

I'm pretty sure that my garage door is not using any kind of rolling code because using a HackRF I'm able to do a replay attack flawlessly and the garage door is opening repeatedly. But using flipper it doesn't work at all.

Some data and pics:

[image]

[image]

Flipper detects it as a fixed code (749083A38) on 433.92 MHz remote with modulation AM270:

[image]

This DIP switch position reading is correct, as you can see in the physical picture:

```
(+): *---*---
(o): --**---*
(-): -*---**-
```

For some reason it detects button 2 (Btn2) but I'm pressing button 1 during the tests. To be honest, I think I never used button 2. The door is opening using button 1.

I know you have probably enough samples. But if you are interested in my samples, I could get them. I could attach the capture of the mastercode, and also raw samples (433.92 MHz - AM270) using different switch positions (all up, all down, all in the middle and the original position described above) pressing button 1 about 3 secs on each. Let me know. I didn't attach them directly because I'm not sure if it is worth for you based on that you already have data on this thread. But if to have another dataset can contribute, let me know. Thanks.

Skorpionm, 2024-08-22:
yes HackRF + SDRSHARP

OscarAkaElvis, 2024-08-22:
Downloaded, .net installed and now running. HackRF connected and selected as source. Any advice about how to configure this?

I guess NFM should be selected. 433.92 freq set. It seems in the right place because some stuff is appearing once I press the garage door button on the controller:

[image]

But no idea about fine tunning. I don't know if some filter or any other configuration should be in place. Actually I don't even know how to record 😆 , trying to find some tutorial to find out how. Any advice is welcome.

OscarAkaElvis, 2024-08-22:
What format do you need? I've read on this thread that somebody saved it as wav and also c16 complex format. So I'm assuming they are the format you are expecting unless you say anything about it.

I can see in the software how to save wav. Not sure about the options to save the wav (stereo, mono), 16 bit PCM, 8 bit PCM or 32 bit IEEE float.  I'm guessing the default stuff (stereo and 16 bit PCM) could be what you are expecting. It seems the "Audio recorder" option can do the trick to get the wav file.

[image]

No idea yet about how to get a c16 format. I started to think that maybe a plugin could be needed to get it. I'll investigate further. Sorry for the re-posting.

Skorpionm, 2024-08-23:
yes it is very similar to the usual AM modulation and the frequency is 433.92. should work with RAW recording. another option is to look at 315, 868.36 are there any transmissions, maybe the remote control (which is unlikely of course) has a wide range and the receiver receives on a different frequency

OscarAkaElvis, 2024-08-23:
Hi. I was not able to get c16 format from SDRSharp. And it seems there is no plugin available to do it, or at least I didn't find the way to do it. So I've decided to capture all from my HackRF portapack. Attached you can find the wav and the c16 files.

[HackRF_captured.zip](https://github.com/user-attachments/files/16730947/HackRF_captured.zip)

Like the last time using flipper, this time using HackRF, I captured the signal doing a long press (2-3 secs) for the four dfferent DIP switches configuration (all up, all down, all in the middle and using the described original configuration).

Hope it helps.

OscarAkaElvis, 2024-08-29:
@Skorpionm , were the data samples valid to you? or didn't have time yet to check? I think the c16 format captured on HackRF portapack should be valid, and also the wav files.

Skorpionm, 2024-08-29:
yes, I looked at them but from them I can conclude that at least raw should work. and I don't understand why it doesn't work. maybe something in the receiver, but without the device on my desk I don't know what to say

OscarAkaElvis, 2024-08-30:
Ohh I can tell you that RAW for some reason is not working. I got all the combinations using different modulations (AM270, AM650 and even the FM ones). And also setting 433.92 freq and also using 422.88 freq (for all modulations) and was not working on any of them, I don't know why. It's quite weird because as I said, using the HackRF, the replay is working like a charm.

What should be the correct modulation on flipper? I guess AM270 because when I use that (not in RAW read, just using read feature) is when the Masterkey is get. Is that mean that is the right modulation? In hackRF when it works, it is set just as NFM. So not sure what this means. What is the mapping from HackRF NFM to flipper modulation?

skotopes, 2024-09-06:
@OscarAkaElvis how many peaks do you see when using spectrum analyzer? 2? 4? more?

OscarAkaElvis, 2024-09-08:
@skotopes it starts as 3 peaks and then there are two remaining on the screen.... this way:

[image]
[image]
[image]

OscarAkaElvis, 2024-09-09:
not sure about the clipping stuff... maybe this control is sending the signal that way intentionally... 

this is the way it is shown on an SDR:

[image]

skotopes, 2024-09-09:
how many peaks? 1 2 4? 

For example 2-fsk:
[image]

4fsk example:
[image]

OscarAkaElvis, 2024-09-09:
I'd say this is 6-fsk, but is a little bit irregular:

[image]

maybe 8?

Another screenshot:

[image]

Normal state without pressing the button to compare as a baseline:

[image]

Skorpionm, 2024-09-09:
no 8 can't be, you're doing something wrong or the transmitter on the remote is really bad

OscarAkaElvis, 2024-09-09:
I think it is really bad because is like 20 years old... but it works. I mean, it opens the garage door always without any problem. Then if it can't be 8, I'd say it's 6.

OscarAkaElvis, 2024-09-09:
I saw you edited my comment 😄 adding this instead of answering on a comment:

> but a "bad" fsk, with a central 433.895 (provided that your SDR is correctly tuned to the frequency) and a deviation of 47 kHz, may well be

Ok. So it seems the center is 433.895 and it contains finally 6 peaks.

Do you need anything else from me? I mean, any more dump or capture from flipper or HackRF?

I know that this could be time consuming to analyze, so take your time. And thank for the patience and the time spent on investigation about these devices. 👍

skotopes, 2024-09-09:
@OscarAkaElvis we need proper record, what you showing to us doesn't looks good. 
What's your setup? What SDR receiver are you using? 

Transceivers used in sub1 usually only support 2fsk,4fsk,mgfsk,OOK,ASK. That allows them to be cheap. In your case most likely it is 2fsk or 4fsk.

OscarAkaElvis, 2024-09-10:
@skotopes, my setup is:

For the flipper dumps:
Flipper with latest momentum007 firmware

For the c16 and wav files:
They were taken directly from the HackRF portapack H2 device (latest mayhem 2.0.1 firmware)

For the waves screenshots:
Same HackRF portapack H2 connected to a Windows host in HackRF mode and using Cubic SDR free software

Maybe this is what is doing this Clemsa devices so special and the reason why they are not yet decoded and included in flipper's stuff. In my case, as I said, it is a 20 years old Mastercode MV1.

This thread was started for a similar device, the Clemsa Mastercode MV12. What about the samples provided by other people? Are they as strange as these, or was there another reason they couldn't be included? Was it due to a lack of samples or unusual patterns, like in mine?

skotopes, 2024-09-10:
@OscarAkaElvis https://www.remote-control-esma.co.uk/clemsa-mastercode-mv-1-remote-control/ is it the one you have?

OscarAkaElvis, 2024-09-10:
yes @skotopes , on my first message of this thread I put the model, and some pics (https://github.com/flipperdevices/flipperzero-firmware/issues/3182#issuecomment-2302447483).

Yeah, it seems the MV1 has only one button (is the one I have), and MV12 has two buttons (the one owned by the author of the thread)

jesusvallejo, 2024-10-10:
Hi, today i was able to replicate my remote using a cc1101 and an esp32 ( https://github.com/mcore1976/cc1101-tool ).
I hope i can clarify some things. The mv12 uses ask modulation, the remote works on 433.92 mhz. 
The remote sends two types of packets. "0" and "1" this packets are about 3.16ms - 3.20ms. The diference between a "0" and a 1 is the time data is being sent. for a "0" you would send about 1.6ms - 1,8ms worth of data and the rest of the packet ( 3.16ms) you would send nothing. for a "1" is about 2.05 ms worth of data. 
the mv12 sends 36 packets. It seems that the first 16 packets are for the key, the folowing 8 are the dip switches and at the end there is some Sync data i presume, or button encoding(mv12 can have 2 buttons).
[image]
In the screenshot you can see the differences between all in + , all in 0 and all in -. 

If all dips are set to +, but the 1, wich is set to - : 
[image]

This should give you enough info to fix the dip switches decoding.

As for the emulation, with a length of 65 microseconds per byte:
65 microsec/byte

aaaaaaaa0000 (3.16ms)
aaaa00000000
aaaa00000000
aaaa00000000
aaaaaaaa0000
aaaa00000000
aaaaaaaa0000
aaaaaaaa0000
aaaaaaaa0000
aaaaaaaa0000
aaaaaaaa0000
aaaa00000000
aaaa00000000
aaaa00000000
aaaa00000000
aaaa00000000
aaaa00000000
aaaa00000000
aaaa00000000
aaaa00000000
aaaa00000000
aaaa00000000
aaaa00000000
aaaa00000000
aaaa00000000
aaaa00000000
aaaa00000000
aaaa00000000
aaaa00000000
aaaa00000000
aaaa00000000
aaaa00000000
aaaaaaaa0000
aaaa00000000
aaaa00000000
aaaa00000000

skotopes, 2024-10-14:
@jesusvallejo @OscarAkaElvis I've received remotes and did some tests. 

Issue with them is quite simple: they use carrier over carrier like in infrared. In other words it got a lot of super short timings.
Our default presets was not designed to handle it, so custom preset required in order to receive it properly.

I'm still not sure how those systems receive signal:
- it may happen that if we ignore 2nd carrier it will work fine(most probably scenario)
- or it also trying to detect that carrier and we'll have to emulate it properly

I'll send those remotes to @Skorpionm for further analysis

OscarAkaElvis, 2025-02-07:
Hi, any updates on this @skotopes? Did you have the opportunity for @Skorpionm to receive them and conduct some tests? thanks.

PR #3187 (2023-11-15): Add Mastercode SubGHz Protocol
# What's new
Resolves #3182

Implements the Sub GHz Protocol for the garage remote Clemsa Mastercode MV12 .

Its a fixed code 433.92 MHz remote with modulation AM270.

Its very similar to Clemsa protocol already implemented, but with some changes:

- Time of the On, Off, pause.
- Length of the key.
- Order of the bits for switches.

# Verification 

The PR includes a raw SUB file to  test the decode and a SUB to test de codification.
Attached are 2 additional SUB to test it.
In the wild you can test it using a remote and using 433.92 MHz and modulation AM270.

#  [... cut]