flipperdevices/flipperzero-firmware #3188: Emulation of Amiibo cards (NTAG215) does not work

bl13pbl03p, 2023-11-02:
### Describe the bug.

The Nintendo Switch does not recognize emulated Amiibo cards from the Flipper, since I updated the firmware to release 0.94.1. Even when a real Amiibo card is read (with the regular read function within NFC) and emulated, the Nintendo Switch will still give the same error: 'This is not an Amiibo.'

Workaround for emulating from real Amiibos: `NFC > Extra Actions > Unlock NTAG/Ultralight > Auth as Ameebo` and then scanning a original Amiibo sortof fixed the issue. 

It seems that something has changed in the latest update that emulates NFC (NTAG125 specifically) differently.

Thank you advance, I am curious about how this bug could be fixed 😊

### Reproduction

**Requirements**:
- Flipper Zero (Release 0.94.1)
- qflipper Desktop application
- Nintendo Switch
- An Amiibo in a .nfc file from [https://github.com/Gioman101/FlipperAmiibo/tree/main](https://github.com/Gioman101/FlipperAmiibo/tree/main)

**Steps:**
1. Download any .nfc file from [https://github.com/Gioman101/FlipperAmiibo/tree/main](https://github.com/Gioman101/FlipperAmiibo/tree/main) to your pc
2. Open the qflipper Desktop application
3. Connect your flipper to the pc
4. Navigate to 'File manager' within in qflipper
5. Navigate to `SD Card > nfc`
6. Drag the donwloaded .nfc file to the File manager
7. Unplug your flipper from the pc
8. On your flipper, click the 'Ok' button
9. Navigate to `NFC > Saved`
10. Choose the .nfc file you downloaded
11. Choose 'Emulate'
12. Take your Nintendo Switch
13. Go to `System Settings > amiibo > `Reset amiibo`
14. Place your flipper on top of the right joy-con of the nintendo switch
15. An error message now appears: `This is not an Amiibo.`

### Target

NFC (NTAG215)

gornekich, 2023-11-02:
Hello @bl13pbl03p . Thanks for the report.

When Switch reads Amiibo tags, it perform authentication with password. Previous firmware version doesn't compare password with saved one and always responds with correct PACK. The value of this PACK is hardcored. That's why the emulation works. Current firmware version compares received password with saved one and responds NACK if the password is incorrect. From protocol implementation point of view, current firmware works correctly, but I can see two problems.

Firstly, Flipper doesn't display information that it didn't find password. We will fix that soon. However, you can see "Unlock" option in card menu, and if you find algorithm to guess password, you can save full dump and emulation will work correctly.

Second problem is with saved dumps from mentioned repository. There are no password and pack saved in that files, and they will not work with new firmware version. Quick workaround here is to add password to 133 and pack to 134 pages. I will come later with better solution soon

u1735067, 2023-11-03:
I got curious on this and NFC formats (and lost hours of time I don't have 🙃), if anyone else is interested:
* https://github.com/flipperdevices/flipperzero-firmware/blob/0.93.0/lib/nfc/protocols/mifare_ultralight.c#L1806 the old hardcoded value behavior mentioned
* https://www.nxp.com/docs/en/data-sheet/NTAG213_215_216.pdf#page=46 the specs / explanation on PWD_AUTH and PACK
  * also https://www.nxp.com/docs/en/data-sheet/NTAG213_215_216.pdf#page=30 on why zeros are returned
* PACK in other projects:
  * tagmo https://github.com/HiddenRamblings/TagMo/blob/96f1636b2acad4dcebb3b93cceebbad9a78ac743/app/src/main/java/com/hiddenramblings/tagmo/nfctech/TagWriter.kt#L281
  * proxmark https://github.com/RfidResearchGroup/proxmark3/blob/v4.17140/armsrc/iso14443a.c#L1285
    * with a human friendly explanation at https://github.com/RfidResearchGroup/proxmark3/pull/1886 (ie. if provided pwd == generated amiibo pwd, return amiibo pack)
* some Amiibo docs https://www.3dbrew.org/wiki/Amiibo

u1735067, 2023-11-03:
You can try https://github.com/u1735067/AmiiboFlipperConverter/blob/patch-1/src/amiiboconvert.py to generate `.nfc` files (with regenerated PWD and PACK) from `.bin` files (one possible source for that is allmiibo).

gornekich, 2023-11-15:
#3207 just got merged. Now flipper will display that it didn't read Ultralight/NTAG cards passwords and you can try to "unlock" them in the menu.
Emulation works as expected, all other issues are connected with emulated files. I will close this issue, if you have any other questions feel free to reopen this issue or create new.

closed 2023-11-15 by gornekich

batstar3, 2023-11-15:
I have the same error, yesterday I could emulate normally, but today it gives me the problem that this is not an amiibo

turbospok, 2023-11-16:
You can try script from here https://github.com/turbospok/Flipper-NTAG215-password-converter which will convert cards and calculate passwords for the existing ones

professorsugoi, 2023-11-22:
> You can try script from here https://github.com/turbospok/Flipper-NTAG215-password-converter which will convert cards and calculate passwords for the existing ones

this worked for me! big thanks (using qFlipper ver 1.3.3)

fanfannnmn, 2023-11-29:
> You can try script from here https://github.com/turbospok/Flipper-NTAG215-password-converter which will convert cards and calculate passwords for the existing ones

It also works well for me! I've incorporated your password generation code into one of the popular project, now it's work again.

olg857, 2024-01-31:
Hello I been trying to go to nfc add manually and then click NTAG215 to read the amiibo from an amiibo app on iPhone and it’s not working I use to be able to do this multiple times to keep making more amiibo made a new UID every time so though it was a new amiibo sadly I don’t have a laptop I only have the flipper zero app on my iPhone