flipperdevices/flipperzero-firmware #3189: NFC Type B card cannot be read

ronny17, 2023-11-02:
### Describe the bug.

With the latest firmware update 0.94.1 my flipperzero cannot read the Type B NFC card

### Reproduction

1. Open NFC app
2. Select read action
3. Apply NFC Type B Card to back

### Target

NFC Type B Card

gsurkov, 2023-11-02:
Could you please give more info on the card?
- Please provide the logs of the reading process (more on gathering logs [here](https://docs.flipper.net/mobile-app/bug-report#710mT)).
- Could it be read with 0.93? If so, it would help if you posted the logs and a save file of it.
- Is it the only NFC-B card in your posession? If not, can other NFC-B cards be read?
- How does it look like? Does it have a manufacturer brand on it or other identifying text?
- Can you read it by other means? (Phone, specialised reader, Proxmark etc - any additional information on the card will be helpful).

ronny17, 2023-11-03:
**Logs of the reading process v 0.94.1**
205433 [D][ST25TBPoller] error during trx: 2
205466 [D][Nfc] FWT Timeout
205506 [D][Iso14443_3bPoller] Unexpected ATTRIB response
205538 [D][Nfc] FWT Timeout
**Logs of the reading process v 0.93**
138639 [I][FuriHalNfc] Deinit OK
138767 [I][FuriHalNfc] Init OK
138772 [D][DolphinState] icounter 101, butthurt 0
With 0.93 the device read the UID of the card.
I have only one NFC-B card (ISO14443B) and i can read it with my pc (uTrust 3700F +Cardpeek) and with my Android smartphone.

gornekich, 2023-11-15:
Hello @ronny17 !
Looks like that Flipper can't parse ATTRIB response. Could you please share data that your Android smartphone or pc can read? The  ATTRIB response is what we need to see

ronny17, 2023-11-22:
**Using nfc-list I get this**
nfc-list uses libnfc 1.8.0
NFC device: PN532Reader opened
1 ISO14443B passive target(s) found:
ISO/IEC 14443-4B (106 kbps) target:
               PUPI: 77  4d  92  42  
   Application Data: 11  00  00  11  
      Protocol Info: 33  81  a3

**Using NXP TagInfo App on my android device i get**
-- FULL SCAN ------------------------------
# Technologies Supported:
ISO/IEC 7816-4 compatible
ISO/IEC 14443-4 (Type B) compatible
ISO/IEC 14443-3 (Type B) compatible
ISO/IEC 14443-2 (Type B) compatible
# Detailed Protocol Information:
ID: 77:4D:92:42
Application Data: 0x11000011
Protocol info: 0x3381A3

gornekich, 2023-12-05:
@ronny17 There are some updates on iso14443-3b poller logic arived in #3264 . Since we can't reproduce this issue, we need your help with testing. Could you please update flipper on latest RC or dev, read your card and provide logs as before.

ronny17, 2023-12-05:
I've updated my flipper to 0.96.1-rc and these are the logs
340384 [D][Nfc] FWT Timeout
340413 [D][Iso14443_3bPoller] Incorrect CID in ATTRIB response: 80

gornekich, 2023-12-06:
Thank you very much. Unfortunately your card doesn't follow iso14443-3b standard. We will make flipper read these type of cards, but in next release. Need some time to think how to implement it better. I will let you know once we make changes.

gornekich, 2023-12-12:
Hello @ronny17 !
I believe #3285 fix the issue. It would be very helpful if you test that PR and confirm that your card is read successfully

ronny17, 2023-12-12:
Now my card is read successfully
Thanks

closed 2023-12-13 by gornekich