flipperdevices/flipperzero-firmware #3203: NFC FeliCa emulation crashes Flipper with null pointer dereference

ntfshard, 2023-11-06:
### Describe the bug.

In 0.94.1 seems we can make a shallow copy of FeliCa cards. But attempt to emulate it shows only animation (NFC tools see nothing, but able to see real card) and attempt to close emulation leads to nullptr dereference crash

### Reproduction

1. save any FeliCa card (e.g. Suica transport card)
2. from idle screen, press down to show NFC menu w/ cards. can be other button, but seems this is default
3. select pre-saved card -> middle button -> Run in app
4. animation starts, led starts blinking (but nothing in NFC tool)
5. press back button - crash

### Logs

```Text
Firmware version: 0.94.1 0.94.1 (e487bcd2 built on 01-11-2023)

>: <ERROR: The device has been lost.>
<DISCONNECTED>
```

gornekich, 2023-11-15:
Hello @ntfshard ! Thanks for the report.
Felica emulation is WIP. The issue will be fixed with #3190 PR

167rgc911, 2023-11-18:
> Hello @ntfshard ! Thanks for the report. Felica emulation is WIP. The issue will be fixed with #3190 PR

I have a custom tree with #3190 merged ... will be testing the Felica feature since everything here (Japan) is Felica!
I captured my ICOCA (commuter pass) using old FW and will be emulating it with the dev FW.

ntfshard, 2023-11-20:
> EDIT/UPDATE: dev + #3190 FW does not crash trying to emulate a Felica card BUT readers can't decode the card!
> 
> by reader, I am using _actual vending machines_ to try to read the emulated card on the Flipper.

I believe for real a vending machine it shouldn't work. I tried on electronic lock which I guess check only something public available data from card

167rgc911, 2023-11-20:
> I believe for real a vending machine it shouldn't work. I tried on electronic lock which I guess check only something public available data from card

My office ID badge is FeliCa. Read it with the Flipper and I just tapped the Flipper on the door access panel in the office.

Got the invalid card beep.

ntfshard, 2023-11-20:
> My office ID badge is FeliCa. Read it with the Flipper and I just tapped the Flipper on the door access panel in the office.
>
> Got the invalid card beep.

In some offices are using suica cards (transport cards) to enter

Jeshii, 2023-11-30:
My FeliCa that I use to get into my office was able to be read and saved but there is no emulate button. My card does not work for transit. Let me know if there is anything I could help test

ntfshard, 2023-12-01:
Tried devel branch, it's working: not crashing, id emulates and door are opens.
Thanks!

167rgc911, 2023-12-01:
0.95.0 OFW does not include FL-3640/3190!

that's why i'm getting the crash again Orz~

gornekich, 2023-12-05:
Now we can emulate FeliCa UID only. We are working to extend FeliCa tags support. 
I will close this issue for now. Please, reopen this Issue if you still have the problem.

closed 2023-12-05 by gornekich