flipperdevices/flipperzero-firmware #3206: cat /dev/ttyACM0 on Linux crashes USB connected flipper with 'furi_check failed'

u0d7i, 2023-11-07:
### Describe the bug.

After connecting Flipper Zero via USB in Linux, and issuing 'cat /dev/ttyACM0' from shell, Flipper Zero crashes with  'furi_check failed' error

### Reproduction

1. Connect Flipper Zero to USB on latest Debian-based Linux machine
2. Locate FZ device
3. Execute 'cat' command on it via default shell: 'cat /dev/ttyACM0' 

### Target

Linux pc 6.1.0-13-amd64 #1 SMP PREEMPT_DYNAMIC Debian 6.1.55-1 (2023-09-29) x86_64 GNU/Linux

### Logs

```Text
[CRASH][CliSrv] furi_check failed
        r0 : 0
        r1 : 20000ef8
        r2 : 20000284
        r3 : 0
        r4 : 0
        r5 : 0
        r6 : 1a0d3
        r7 : 13870
        r8 : 0
        r9 : 20000ecc
        r10 : 1a0e0
        r11 : 20031330
        lr : 8010cff
        stack watermark: 3420
             heap total: 186024
              heap free: 79984
         heap watermark: 32552
        core2: not faulted
Rebooting system.
```

### Anything else?

Unable to reproduce on Mac OS X, but the same is on several Linux hosts, so, may be Linux specific.

u0d7i, 2023-11-07:
Firmware version:       0.94.1

hedger, 2023-11-08:
@doomwastaken please check and confirm on 94.1 and dev, with stack traces on crash

doomwastaken, 2023-11-14:
We were able to reproduce it, will update once we have details

hedger, 2023-11-14:
```

Thread 3 (Thread 537078888 (Name: CliSrv, State: Running)):
#0  0x080100d6 in __furi_crash_implementation () at furi/core/check.c:164
#1  0x08010d5a in pvPortMalloc (xWantedSize=<optimized out>) at furi/core/memmgr_heap.c:485
#2  0x08010728 in realloc (ptr=ptr@entry=0x2001c4a0, size=size@entry=71137) at furi/core/memmgr.c:26
#3  0x08011b24 in m_str1ng_fit2size (v=v@entry=0x20003fd0, size_alloc=size_alloc@entry=47425) at lib/mlib/m-string.h:290
#4  0x0801226a in string_push_back (c=46 '.', v=0x20003fd0) at lib/mlib/m-string.h:474
#5  furi_string_push_back (v=0x20003fd0, c=<optimized out>) at furi/core/string.c:148
#6  0x08060600 in cli_process_input (cli=cli@entry=0x20003f30) at applications/services/cli/cli.c:360
#7  0x080607e0 in cli_srv (p=<optimized out>) at applications/services/cli/cli.c:471
#8  0x0801271a in furi_thread_body (context=0x20003e00) at furi/core/thread.c:92
#9  0x080126c6 in furi_thread_catch () at furi/core/thread.c:63
Backtrace stopped: previous frame identical to this frame (corrupt stack?)
```

Flipper runs out of memory, because `cat`, for some reason, is spamming data into the VCP.

doomwastaken, 2024-02-27:
Issue was fixed in #3358

closed 2024-02-27 by doomwastaken