flipperdevices/flipperzero-firmware #3208: Incorrect version of Protobuf in RPC if non-release branch is built

Programistich, 2023-11-08:
### Describe the bug.

Now in PR https://github.com/flipperdevices/flipperzero-firmware/pull/3180 we include new protobuf update, but not create new tag, which causes the flipper build system to use the latest release tag rather than the current one in the pull request.

### Reproduction

1. Create new branch with changes in protobuf
2. Create new branch with changes in Flipper FW with new Protobuf API and create Pull Request
3. Build PR and see old version protobuf

hedger, 2023-11-08:
You can add a local tag for testing purposes on that submodule.

skotopes, 2024-01-10:
Feature )

closed 2024-01-10 by skotopes