flipperdevices/flipperzero-firmware #3209: NFC: talk to type 4 cards

bettse, 2023-11-09:
### Description of the feature you're suggesting.

I'm trying to talk to a seos (14443a-4 javacard) using `nfc_poller_trx` and getting
```
3373360 |    3391888 | Rdr |00  a4  04  00  0a  a0  00  00  04  40  00  01  01  00  01  00           |  !! | 
```

but testing pre-refactor what I want is 
```
3955824 |    3977872 | Rdr |02  00  a4  04  00  0a  a0  00  00  04  40  00  01  01  00  01  00  fe   |     | 
            |            |     |6b                                                                       |  ok | 
```
I believe this is framing and I should use something like `iso14443_4a_poller_send_block`, but this isn't exposed.  Am I doing this wrong, or can this be exposed?

skotopes, 2023-11-09:
@gornekich FYI

gornekich, 2023-11-09:
Hello @bettse !
I see the problem. I am working on API improvements right now and I will expose some functions, including iso14443_4a_poller_send_block(). Hopefully I will make PR today

gornekich, 2023-11-15:
#3214 just merged. I believe it resolves this issue. Now internal poller API is exposed.

I will close this issue for now. If you have any questions regarding this question, please reopen it.

closed 2023-11-15 by gornekich