flipperdevices/flipperzero-firmware #3212: NFC: SLIX no longer works

exussum12, 2023-11-12:
### Describe the bug.

Both previously saved and trying to read a new card fail on the current firmware. 

Flipper displays emulating iso15694-3 (unknown) when trying to emulate a previously working file 

### Reproduction

See above

gornekich, 2023-11-15:
Hello @exussum12 !
Could you please provide your file dump? It will help to debug the issue.

exussum12, 2023-11-15:
Hey I can send it to someone on Discord or via email - whats best to send?

gornekich, 2023-11-15:
Please send me in Discord. My nick is gornekich

gornekich, 2023-12-05:
Still working on this issue. Planning to include support of all older file versions in next release 0.97

tdimov93, 2023-12-06:
hey @gornekich, I'm also noticing a couple of issues related to SLIX cards (SLIX2 in my case):

- after reading a tag if I click on _More -> Info_, the Flipper reboots and shows a "NULL pointer dereference" message upon startup
- trying to click on a saved SLIX2 file results in a "Cannot load key file", even if it was created by the flipper itself a second or two ago

I saw this running the official firmware v0.95.0

Please let me know if you'd like me to open a separate issue for this or if you think it might be related to the original issue (or handled as part of it).
I'd be happy to assist with more information if necessary

gornekich, 2023-12-06:
Hi @tdimov93 
Thanks for information. I will work on SLIX support improvements soon and will ask you to reproduce this issues after I am done. No need in opening another issue, let's discuss it here.

tdimov93, 2023-12-06:
great, thanks!

Unfortunately I'll only have access to the problematic tag for another 6 days. I have the saved file that refuses to open so I'll be able to reproduce the 2nd issue after that without problems but the first issue would be impossible for me to reproduce after Tuesday. Let me know if I can help with grabbing some logs/traces for you that might help you identify or reproduce it on your end as I don't imagine you'll get around to working on it this week

exussum12, 2023-12-06:
I have a bunch of tags I can scan if needed

tdimov93, 2023-12-06:
Do you have any password-protected ones? I just received a pack of 20 fresh SLIX2 tags and I didn't manage to reproduce the either of the 2 issues I mentioned with them. It would seem the problems arise when the flipper fails to read the password-protected data section of the tag

exussum12, 2023-12-06:
Yeah I have passworded ones. The old firmware could unlock

tdimov93, 2023-12-06:
Which version are you referring to when you say old firmware? 0.94.1?
Btw is there a way to grab the commands and responses in order to debug NFC with the flipper zero without an external tool? Haven't been able to find info on that

exussum12, 2023-12-06:
0.93 I think was the latest I could use.

exussum12, 2024-01-01:
Just tried 0.97.1 and this is sill broken

tdimov93, 2024-01-17:
Same. Regarding the issue I described, I compared the corrupt NFC file dump with a regular empty SLIX2 tag dump and saw that the problematic one has an empty `Security Status: ` section. When I copied the section from the empty tag file over manually and saved it (it's a bunch of 0s), the flipper was able to parse it successfully. My guess is that given the original tag is likely password-protected, the flipper fails to read that section and ends up writing an empty data array instead of populating it with question marks for example and leaves the file corrupt.

gornekich, 2024-06-19:
Hi @exussum12 . Could you please try lates dev again?

skotopes, 2024-07-07:
Please check latest release, also use `Capabilities: AcceptAllPasswords` if you bypassing auth. Reopen if problem persist.

closed 2024-07-07 by skotopes