flipperdevices/flipperzero-firmware #3213: In level 2

Danucosukosuko, 2023-11-12:
### Describe the bug.

My flipper is in the level 2 and shows the animations of level 1

### Reproduction

Automatic

hedger, 2023-11-12:
Dies it show animations from level 1 only? Or both 1&2?

Danucosukosuko, 2023-11-12:
Only the 1 Level

Danucosukosuko, 2023-11-12:
I tested in 0.90.1 FW and nothing, but in 0.94.1 shows the L1 Animations

Danucosukosuko, 2023-11-12:
I tested in 0.90.1 FW and shows both Animations, L1 & L2, but in 0.94.1 shows the L1 Animations

doomwastaken, 2023-11-13:
Unable to reproduce, tested on lvl2, lvl3 devices in Ok / Angry mood

Danucosukosuko, 2023-11-13:
This morning I finally got a Lvl2 one. I think you should make more of them come out of lvl2 than 1

doomwastaken, 2023-11-14:
You can cycle animations by turning device on and off, that way you can be sure that they are working correctly. As for animations, here is the list
[image]

I will close this issue, reopen it if issue persists

closed 2023-11-14 by doomwastaken

Danucosukosuko, 2023-11-14:
Ok